yaacov/node-modbus-serial #2: Need to buffer serial data.

brycheiniog, 2015-12-22:
Hi,

I just tried running v3.2.5 of the package in ModbusRTU mode against this Modbus Simulator: http://www.plcsimulator.org/

When I try and run the simple logger example, I get loads of errors about the Data Length not being what is expected (See line 192 of index.js). 

This is happening because the serialport module is passing you data as it gets it from the far end and not waiting. 

One option would be to configure the serialport module to wait for a delimiter before triggering the 'data' event, but from a quick look at the modbus protocol I don't think that is possible. If that is the case then I think the correct solution is to buffer the data up until you have the expected length and then attempt to process it.

yaacov, 2015-12-23:
hi,

thank you for the issue !

please check the git version, i added a bufferd serial port:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/rtubufferedport.js
https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger-buffered-rtu.js

happy solstice.

brycheiniog, 2015-12-23:
Great, that has fixed the issue I was seeing. 

Thanks!

closed 2015-12-23 by brycheiniog

yaacov, 2015-12-23:
:-)

brycheiniog, 2015-12-23:
Are you going to release a new version of the package?

Thanks

yaacov, 2015-12-23:
v3.3.5 !