yaacov/node-modbus-serial #5: Inconsistent error handling

connium, 2016-01-27:
Hi Yaacov,
thanks for providing this module :)

While reading the code I wonder at how the error handling is done.

In index.js the on-data-handler checks if the emitted data matches the expectations (address, function code, length, ...). If any check fails, the callback is invoked with an error.

This works well for the TestPort and probably for the TcpPort as well (can't test it yet). But for the other ports the error handling is done inside the ports and thus only valid data is emitted to the index.js...
To me it feels correct to execute the data check inside the ports, as they are the only ones knowing how to process the specific messages (check and cut off headers, ...).

What's your opinion about this?

Best regards

FYI: I'm working on modbus exception handling.

yaacov, 2016-01-27:
If it follows the Zen:
    Beautiful is better than ugly.
    Explicit is better than implicit.
    Simple is better than complex.
    ...
It's OK by me :-)

yaacov, 2016-01-28:
@connium I'm a little overwhelmed right now with other stuff, 
if you are looking into the code for the exception handling ... can you review:
https://github.com/yaacov/node-modbus-serial/pull/4
1. It's a little rough around the edges.
2. I do not have any controller that implement FC15, so I can not check it in the wild :-)

connium, 2016-01-28:
Sure. I'll check #4 tomorrow morning.

yaacov, 2016-01-30:
I looked at the error handling is done inside the ports ...
It is very hard to know if you have a partial-message / message-for-someone-else or an actual error.
I do not see a simple way to do this ... If you can, this will be very nice :-) 

p.s.
if you can't do it in a simple way, leave it. 
For me, it is better to have a clear maintainable code, then to handle errors for the None Modbus Complaint ports (e.g. c701, telnet and buffered-serial).

yaacov, 2016-01-31:
I've started to handle modbus exceptions, see code.
1. Currently, it only output the exception code, and not the exception string, TODO  1.
2. The buffered ports do not support this yet, TODO 2 :-)

connium, 2016-02-01:
I will take a look at your code as soon as I have time to do so.

Here some hints (we are working on in our branch but we didn't commit to https://github.com/meteocontrol/node-modbus-serial yet):
- extract the checksum-calculation to utils/crc16 (currently it is copied in each port)
- refactor the on-data-handler in index js to be a method ModbusRTU.prototype. This will largely increase testability.
- streamline data checking of the ports and extract it to utils/checkResponse
- rename the variables that refer to the slave ID to **unitId** as this seems to be the official naming (your public API already matches this naming pattern)

BTW: We use Modbus buffered-serial (as we don't want to handle each incoming byte by ourselves) and Modbus TCP. Why do you call buffered-serial none-Modbus-compliant? It's just a parser on top of serial port, isn't it?

Thanks for your great work you do with this project!

yaacov, 2016-02-01:
Just looked at your branch, Great ! 
1. liked the javaDoc fixes :+1: 
2. Please re-base to current, because it will make it easier to merge.
3. I've added a buffer_bit.js file to api directory, it's place is in the new utils directory :-)

yaacov, 2016-02-01:
@connium : buffered-serial none-Modbus-compliant?
The modbus serial line should not be buffered:  
http://www.modbus.org/docs/Modbus_over_serial_line_V1.pdf
see section 2.5.1.1 : MODBUS Message RTU Framing

[ I also worked with controllers that did not respect this, this is why we have the buffered serial :-) ]

yaacov, 2016-02-10:
@connium I want to add your changes to the javaDoc and move of utils to a utils directory.
can you make a PR for that ?

connium, 2016-02-11:
Do you favor a name for the utils directory? Something like lib, util, utils?

yaacov, 2016-02-11:
no i do not have a convention yet :-) use what you think is the best convention you know.

connium, 2016-02-11:
Will close this issue for now and provide you with some pull requests which address the topics mentioned above.

closed 2016-02-11 by connium

yaacov, 2016-02-11:
:-)