yaacov/node-modbus-serial #13: pulling out crc16 has broken response check

dom-white, 2016-02-11:
The new single crc16 function works for appending to requests, but fails when checking responses as it no longer skips the last two bytes i.e. the crc of the response packet.

The crc version when checking repsonses included this extra line, then referenced 'length' for the for loop rather than buf.length

var length = buf.length - 2;

dom-white, 2016-02-11:
Could be fixed by passing in a slice of the buffer as per the request side
crc16(buf.slice(0, -2))
rather than
crc16(buf)
in the ports checkData method

dom-white, 2016-02-11:
I've sent a pull request which I think fixes the issue.
It was fine on the test port, but not on the serial,tcp,udp ports

dom-white, 2016-02-11:
Closing as pull request now merged in.

closed 2016-02-11 by dom-white

connium, 2016-02-11:
Obviously missed some parts when cherry-picking from another branch. Damn!
Thanks @dom-white for spotting and fixing this!

yaacov, 2016-02-11:
:+1:  sorry for not checking before merging.