yaacov/node-modbus-serial #17: Use semantic versioning

connium, 2016-02-20:
Currently it is not very obvious to me, how version numbers are set. E.g. with the latest commit, which simply added some tests, the major version number was increased.
I suggest to use semantic versioning (http://semver.org/).

Given a version number MAJOR.MINOR.PATCH, increment the:
1. MAJOR version when you make incompatible API changes,
2. MINOR version when you add functionality in a backwards-compatible manner, and
3. PATCH version when you make backwards-compatible bug fixes.

This will help users of this library to determine what happened between several versions. Especially for professional users this is important, as they need to know when breaking changes are made to the API.

As the current version is 4.0.0 I think this would be an excellent version to start with the new scheme.

yaacov, 2016-02-20:
OK ! semantic versioning is good :+1:  , 4.0.0 is a good place to start use it.

P.S 
Until 4.0.0 we used the happiness versioning (tm) system. It is a good system for projects where I'm the only user :-)
1. MAJOR version when I was extremely happy with the change,
2. MINOR version when I was just happy with the change, and
3. PATCH version when it was something that we will need to fix in the future (until happiness achieved) .

closed 2016-02-20 by yaacov

connium, 2016-02-20:
I wasn't aware of the happiness versioning (tm). It sounds like it is an extremly lightweight system and I'll think of it when I have to choose a decent versioning system for future use ;)

You made my day :joy: