yaacov/node-modbus-serial #19: Error Handling on connectTCP broken

numbnut, 2016-02-25:
It is not possible to handle socket errors on a connectTCP call.

The suggested behavior of the example would be to print out "error handled".
Instead a Unhandled Error Event occurs.
Precondition: The port 4444 is closed!

`// create an empty modbus client
var Modbus = require('./index.js')
var client = new Modbus();

// open connection to a tcp line
client.connectTCP("localhost", {port: 4444}, (error) => {
  if(error){
    console.log('error handled')
  }
  console.log('connected')
});
`

In my opinion the problem is the call of net.Socket.connect(port, ip, eventListener) in tcpport.js in line 57. The Docu of net.Socket expects a eventListener but not a callback!
The callback passed to connectTCP(...) will not be called in case of an error.

If I can do anything else to assist fixing this issue just respond. Thanks.

yaacov, 2016-02-25:
Thanks for the issue !
how do you suggest to address this ?

a. make the documentation clear that in the case of tcp, telnet and c701 the last argument is event-listener and not a callback ?
b. make the callback tol be called from a listener we create in the open method ?
c. other option ?

can you make a pull request with the option you like best ?

yaacov, 2016-02-25:
This should solve this issue:
https://github.com/yaacov/node-modbus-serial/commit/16387c5d564cc9ace417b6a5257a36b7110215aa

yaacov, 2016-02-28:
fixed by: https://github.com/yaacov/node-modbus-serial/commit/60250fb5daa3b00882b3ac44a6f510a81aa10da7

closed 2016-02-28 by yaacov

numbnut, 2016-04-06:
Thank you very much!