yaacov/node-modbus-serial #22: about timeout

fly19890211, 2016-05-17:
Hello, thanks for this module :)
and i'm using connectRTUBuffered, is there timeout for this api? 

Thanks

yaacov, 2016-05-17:
Hi, thanks, it use the serialport module, timeout is a property of the serial port ...
https://github.com/voodootikigod/node-serialport
can you ask there ?

fly19890211, 2016-05-18:
Hi,
ok. i will do that, thanks.

yaacov, 2016-05-20:
Hi, did you found how solve the problem ?

fly19890211, 2016-05-28:
hello, thanks for replying.
actually, i can't find the how to solve the problem...

yaacov, 2016-05-28:
can you explain what is your use case / what you are trying to do ?
maybe their is a way to solve you problem, that does not involve timeouts ?

fly19890211, 2016-06-01:
my case is i use it for reading inverters of solar energy station. and sometimes the inverter doesn't respond, so i need to timeout to escape the connectRTUBuffered.
Thank you.

yaacov, 2016-06-01:
hi, thanks for answering

take a look at:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/watchdog.js

1. it's tcp but it should work on rtu.
2. we have a new `close` method :tada: , so you can also close the connection and reconnect if you want.

yaacov, 2016-06-09:
Fixed by: https://github.com/yaacov/node-modbus-serial/pull/26

closed 2016-06-09 by yaacov

fly19890211, 2016-06-09:
Thanks, really appreciate for this updating. 
can you please update the Readme.md about how to use timeout.

yaacov, 2016-06-10:
> can you please update the Readme.md about how to use timeout.

https://github.com/yaacov/node-modbus-serial/blob/master/examples/watchdog.js