yaacov/node-modbus-serial #25: DC error  CRC error:data:undefined

robinxc, 2016-05-31:
SOURCE:
 var ModbusRTU = require("modbus-serial");
 var client = new ModbusRTU();
  client.connectRTU("/dev/ttyS1", {baudrate: 9600});
 client.setID(1);
setInterval(function() { 
	client.readHoldingRegisters(1, 2, function(err, data) {
		debugger;
        console.log(err+':data:'+JSON.stringify(data) );
    });
	   
}, 7000);
===================
ERR:
CRC error:data:undefined
Data length error, expected 9 got 1:data:undefined
CRC error:data:undefined
Data length error, expected 9 got 1:data:undefined

closed 2016-05-31 by robinxc

yaacov, 2016-05-31:
What was the problem ? how it was solved ?

pagalasoujanya, 2016-11-15:
i'm also getting the same error can you tell me how to solve that error

yaacov, 2016-11-15:
Hi,

can you try: `connectRTUBuffered` instead of `connectRTU`

pagalasoujanya, 2016-11-16:
i connected an instrument reading value from it through the modbus. instead connectRTU i tried connectRTUBuffered but it not printing any value

pagalasoujanya, 2016-11-16:
Port Not Open:data:undefined

getting the above error

yaacov, 2016-11-16:
Hi,

Try doing it the long way, and see where you get an error:

```
// open a serial port
var SerialPort = require("serialport");
var serialPort = new SerialPort("/dev/ttyUSB0", {baudrate: 9600, autoOpen: false});
```
Check the serial port options:
dataBits, stopBits, parity ...
https://github.com/EmergingTechnologyAdvisors/node-serialport#module_serialport--SerialPort..openOptions

then do:
```
// create a modbus client using the serial port
var ModbusRTU = require("modbus-serial");
// or for bufferd port
// var ModbusRTU = require("modbus-serial/ports/rtubufferedport");
var client = new ModbusRTU(serialPort);

// open connection to a serial port
client.open();
```
continue with your program,
and now see what happen.

p.s.
make the shortest script you can, that has this problem and post it here, maybe we will see something ... :-)

yaacov, 2016-11-16:
can you check the module versions you use ?
`npm ls serialport`
and
`npm ls modbus-serial`

pagalasoujanya, 2016-11-16:
serialport 4.0.6  and  4.5.1

pagalasoujanya, 2016-11-16:
not printing any value

pagalasoujanya, 2016-11-16:
i tried with the example code, which is there in the modbus-serial package in example folder only write function is printing on the console but it read function is not calling.

yaacov, 2016-11-16:
> i tried with the example code

make the shortest script you can, that has this problem and post it here, maybe we will see something ... :-)

pagalasoujanya, 2016-11-16:
// create an empty modbus client 
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();
 
// open connection to a serial port 
client.connectRTU("/dev/ttyUSB0", {baudrate: 9600});

 

// read the values of 2 registers starting at address 1 
// on device number 1. and log the values to the console. 
setInterval(function() {
    client.readHoldingRegisters(1023, 1, function(err, data) {
	console.log("error",err);
if(!err){
        console.log(data.data);
}	//console.log('abc');
    });
}, 1000);

pagalasoujanya, 2016-11-16:
error Data length error, expected 7 got 1
error Data length error, expected 7 got 1
 getting the above error
my device starting address is 1023

yaacov, 2016-11-16:
1. check the bouderate, stopbits, databits and parity of you device
2. make sure is holding registers
3. check the address, some devices use different adressing conventions:
https://en.wikipedia.org/wiki/Modbus#Coil.2C_discrete_input.2C_input_register.2C_holding_register_numbers_and_addresses
4. is your device connected to "/dev/ttyUSB0" ?
5. try this code, 
    try change the serial settings for your device (e.g. boude, parity ...)
    try adress +400000
    and see what the output is.
6. do you have another way to see if your device is connected correctly ?

```
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

// check that the boud, parity, bits are as the device expect
client.connectRTUBuffered("/dev/ttyUSB0", {baudrate: 9600}, read);

function read() {
    client.setID(1);

    // read from device number 1
    // 1 holding register starting at address 1023
    client.readHoldingRegisters(1023, 1)
        .then(function(d) {
            console.log("Recive:", d); })
        .catch(console.log);
}
```

pagalasoujanya, 2016-11-16:
getting the output like this

Recive: { data: [ 1 ], buffer: <Buffer 00 01> }

pagalasoujanya, 2016-11-16:
Thank you its working with out any error

pagalasoujanya, 2016-11-16:
buffer: <Buffer 00 01> what is mean by this buffer what is printing on the output

yaacov, 2016-11-16:
> buffer: what is mean by this buffer what is printing on the output

?
I do not see what you see :-) what are you trying to say ?

pagalasoujanya, 2016-11-16:
thank you im getting the output

yaacov, 2016-11-16:
> thank you im getting the output

so what was the proble ?

yaacov, 2016-11-16:
@pagalasoujanya 
did you solve the problem ?
how ?

pagalasoujanya, 2016-11-17:
solved i tried with the client.ModbusRTUBuffered( With all the device configurations)

yaacov, 2016-11-17:
p.s
Use `connectRTUBuffered` when the device does not answer in one data frame.

The Modbus protocol specify timing for sending one data frame [1] 
If the device does not follow the protocol and sent the data in fragments of frames, we collect them in a **buffer** and use the buffer to build complete frames we can parse.

[1] https://en.wikipedia.org/wiki/Modbus#Frame_format

298christian, 2021-02-04:
> Hi,
> 
> can you try: `connectRTUBuffered` instead of `connectRTU`

Thanks