yaacov/node-modbus-serial #31: implement transaction identifier for Modbus/TCP

valfis86, 2016-06-13:
Modbus/TCP messages contain a "transaction identifier". This field can be used to correlate a modbus response to the request message. 

While sending multiple modbus/TCP requests to a client I faced some issues regarding the length of the expected modbus response ("Data length error, expected ..."). It seems like that you use one field to store the expected length of the answer when sending a new request. This field is overwritten each time a message is send. This leads to errors when sending multiple simultaneous requests with different lengths. 

I was sending a readInputRegisters message directly after sending a writeRegisters message.

yaacov, 2016-06-13:
Thanks, I will look at it this weekend.
If you can make a pull request, It will be faster.

yaacov, 2016-07-08:
fixed by #33

closed 2016-07-08 by yaacov