yaacov/node-modbus-serial #34: Packet length check does not work when reading one coil

cgoebeler, 2016-07-26:
When trying to read a single coild value from modbus via tcp I receive a "Data length error" originating at the second length check (version 4.4.3 index.js: line 256).

It works as soon as I read multiple coils.

If I comment out the length check the package itself is decoded just fine.

I'm tested this with a Codesys Runtime on a Raspberry Pi. Unfortunately I don't have this modbus slave here right now, so I can't paste the actual package content.

yaacov, 2016-07-26:
Hi, thanks :+1: 

I will not have time to check it until the weekend ... 
If you can find the problem, and make a pull request with a fix It will be great.

cc @meteocontrol

cgoebeler, 2016-07-26:
Alright, I think I'll have the modbus slave back here later today. Will fix it.

yaacov, 2016-07-26:
> When trying to read a single coild value from modbus via tcp

What command do you use to read one coil ?

cgoebeler, 2016-07-28:
I got my modbus slave today and after reinstalling the node module in version 4.4.3 I can not reproduce the issue anymore, also the length check is at a completely different source code location now I seem to have use an older version. Sorry for the inconvenience.

closed 2016-07-28 by cgoebeler

yaacov, 2016-07-28:
NP :-)
It's good to know new versions are improving ...
We have v4.5.0 now that support the new serial port API