yaacov/node-modbus-serial #38: Problems with serial sequenical reads

hardillb, 2016-08-22:
I'm getting `Data length error` messages when trying to make multiple reads via serial.

I have a really simple setup, a rPi with a USB FTDI RS485 adapter linked to a single modbus slave.

It either fails on the second or third read, the first always works.

The code is pretty simple:

```
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

client.connectRTU('/dev/ttyUSB0', {baudrate:38400, dataBits: 8, parity: 'none', stopBits: 1}, read1);
var i;

function read1() {
  if (!i) {
     client.setID(1);
     i = 1;
  }
  client.readHoldingRegisters(2009,2,function(err,d){
    console.log("read1");
    if (err) {
      console.log("read1 error");
      console.log(err);
    } else {
      console.log(d);
      read2();
    }
  });
}

function read2() {
  client.readHoldingRegisters(2010,1,function(err,d){
    console.log("read2");
    if (err) {
      console.log("read2 error");
      console.log(err);
    } else {
      console.log(d);
      read3();
    }
  });
}

function read3() {
  client.readHoldingRegisters(2011,1,function(err,d){
    console.log("read3");
    if (err) {
      console.log("read3 error");
      console.log(err);
    } else {
      console.log(d);
      setTimeout(read1,3000);
    }
  });
}
```

Any ideas what's going on?

yaacov, 2016-08-22:
Hi, thanks

can you try using the buffered serial port ?
e.g.
`client.connectRTUBuffered('/dev/ttyUSB0', {baudrate:38400, dataBits: 8, parity: 'none', stopBits: 1}, read1);`
and see if the error changes ?

hardillb, 2016-08-22:
That seams to have fixed things

Thanks

closed 2016-08-22 by hardillb

yaacov, 2016-08-22:
Well some slaves do not follow the modbus timing rules :-( , 
The buffer-RTU use a buffer to collect a fragmented message from the slave and build one valid message .
Another way to solve it will be to add a very small delay between readings ( like you did in read3, but use less time )