yaacov/node-modbus-serial #39: Parallel ModbusTCP

jovermier, 2016-10-27:
I created an expressjs route that when called does:

Simplified version:

app.get('/modbus', function(req, res){
let client = new ModbusRTU();
 
// open connection to a tcp line 
client.connectTCP("192.168.1.42", function() {
     client.readHoldingRegisters(1, 1, function(err, data) {
	if (err) {
		res.send(err);
	} else{
		res.send(data);
	}
   });
 });
}
I have a frontend app that polls this service. Everything works perfectly until I open up two instances of the web page. When the service has multiple connections to it there seems to be an issue. The time to get the response goes up an order of magnitude.

Another way I noticed this issue is when using asyncjs and reading a bunch of modbus registers in parallel.

yaacov, 2016-10-28:
Hi, thanks for noticing that !

In the real app, do you have one global connected client or you create and connect each time a page is called ?

If you create a client and connect each time you get a http request:
a. try to close the connection after using it - `client.close()`
b. try to create a global `client` connect it, and after connection is made, start listening to http requests and use this already connected client.

yaacov, 2016-10-28:
> Everything works perfectly until I open up two instances of the web page.

Can your IoT device handle two open tcp sockets ?

jovermier, 2016-10-28:
I tried `client.close()` which solved the issue. Thanks!

closed 2016-10-28 by jovermier

yaacov, 2016-10-28:
:-)