yaacov/node-modbus-serial #42: TypeError: Cannot read property 'isOpen' of undefined

biancode, 2017-01-06:
Hi,

in code I've that:
if (node.client && node.client.isOpen && node.client.isOpen())
to check for close or not

but sometimes: 
TypeError: Cannot read property 'isOpen' of undefined

Idea what's wrong?

yaacov, 2017-01-06:
i do not think I ever implemented `isOpen` for `ModbusRTU` only for the ports ( e.g. the `._port ` ) ...
but then why does it do work sometimes ... !?

on the other hand I do check for `obj._port.isOpen()` before each send command, so if the `obj._port` is undefined (never opened , or closed) and you try to send something, you can get this error.

yaacov, 2017-01-29:
fixed by https://github.com/yaacov/node-modbus-serial/pull/49

closed 2017-01-29 by yaacov