yaacov/node-modbus-serial #43: Modbus TCP

Yarisrespect, 2017-01-12:
How to implement Modbus TCP protocol using modbus-serial?
Its has modbus over tcp, etc.

yaacov, 2017-01-12:
```
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

// open connection to a tcp line
client.connectTCP("192.168.1.42", run);

// read the values of 10 registers starting at address 0
// on device number 1. and log the values to the console.
function run() {
    client.setID(1);

    client.readInputRegisters(0, 10)
        .then(console.log)
        .then(run);
}
```

yaacov, 2017-01-12:
@Yarisrespect does that answer your question ?

Yarisrespect, 2017-01-12:
I saw that, but it is modbus over tcp I am talking about Modbus TCP wich is without CRC

Yarisrespect, 2017-01-12:
Modbus RTU over TCP

Simply put, this is a Modbus RTU message transmitted with a TCP/IP wrapper and sent over a network instead of serial lines. The Server does not have a SlaveID since it uses an IP Address instead.

Modbus TCP

A Modbus Messaging Implementation Guide provided by Schneider Automation outlines a modified protocol specifically for use over TCP/IP. The official Modbus specification can be found at www.modbus.org/specs.php . The main differences between Modbus RTU and Modbus TCP are outlined here.

from http://www.simplymodbus.ca/TCP.htm

yaacov, 2017-01-12:
ok :-)

`.connectTCP` - Modbus TCP/IP or Modbus TCP — This is a Modbus variant used for communications over TCP/IP networks, connecting over port 502.

`.connectTelnet` - Wraps the Modbus RTU and send it over TCP using telnet.
`.connectC701` - Wraps the Modbus RTU and send it over UDP using the C701 proprietary protocol.

yaacov, 2017-01-12:
@Yarisrespect do you need Modbus TCP or Modbus RTU/IP ?

Modbus TCP , we have using `.connectTCP`
Modbus RTU/IP , we do not have. but the `telnet` variant sounds very similar.

Yarisrespect, 2017-01-12:
I meen that this lib sends whole RTU.
But in some cases you need send only PDU.

And one more question. Why do _transactionId in ModbusRTU object always equals 1?

yaacov, 2017-01-12:
> I meen that this lib sends whole RTU.
> But in some cases you need send only PDU.

no, the tcp variant , ascii variant and rtu variant send different frame formats,
they will not work o/w :-)

> Why do transaction Id in ModbusRTU object always equals 1?

a. only the tcp frame format has a transaction identifier ( see answer to question 1 ) 
b. it is not, transaction identifier is automatically set and is different for each transaction.

Yarisrespect, 2017-01-12:
I am sorry. You have a point there.

yaacov, 2017-01-12:
@Yarisrespect you can look at the tests:
this transaction has an id of 1 ( see the second byte of frame ):
https://github.com/yaacov/node-modbus-serial/blob/master/test/ports/tcpport.test.js#L73
and the next transaction has an id of 2 
https://github.com/yaacov/node-modbus-serial/blob/master/test/ports/tcpport.test.js#L82

Yarisrespect, 2017-01-12:
Thank you a lot) If I want to send CRC with PDU togeher over TCP I should just implement new port that would send it. Right?

yaacov, 2017-01-12:
i think the telnet port is doing exactly what you want :-)

yaacov, 2017-01-12:
> Thank you a lot) If I want to send CRC with PDU togeher over TCP I should just implement new port that would send it. Right?

@Yarisrespect 
a. yes
b. I think that the telnet port is doing just that, see comment above ^^

closed 2017-01-12 by yaacov