yaacov/node-modbus-serial #44: Is it possible to enhance this package with slave/server

biancode, 2017-01-13:
Is it possible to enhance this package with slave/server, please?

http://www.rtaautomation.com/technologies/modbus-tcpip/

yaacov, 2017-01-13:
> Is it possible to enhance this package with slave/server

a. yes :-)
b. what is the use case ?
c here is mock server, is this what you have in mind ?
```
// create an empty modbus server
var ModbusRTU = require("modbus-serial");
var server = new ModbusRTU();

// open connection to a tcp line
// and register some response functions 
client.serveTCP({url: "0.0.0.0", port: 502} , { 
    readInputRegisters: function(address, length) { return [42; ]; },
    writeCoil: function(address, state) { console.log("writing coil"); } 
});

```

biancode, 2017-01-13:
a. great!
b. https://github.com/biancode/node-red-contrib-modbus/blob/master/src/modbus-server.js
want to use only modbus-serial in node-red package
c. looks good and also see https://github.com/biancode/node-red-contrib-modbus/blob/master/src/modbus-server.js from line 111

yaacov, 2017-01-29:
fixed by #52

see example:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/server.js

closed 2017-01-29 by yaacov

biancode, 2017-01-30:
cool - I'll start testing next week