yaacov/node-modbus-serial #45: TCP can not handle async parallel unit requests

biancode, 2017-01-17:
I tried to send multiple commands to different unit-id's per cycle via TCP, but it brings errors.
Each unit gets only one command per cycle, but we sent four at the same time.

You can check that with http://www.plcsimulator.org/ it provides multiple units, please.

[image]

biancode, 2017-01-17:
just to handle with multiple tcp connections

closed 2017-01-17 by biancode

inpras, 2017-03-06:
"just to handle with multiple tcp connections"
What does this mean? does this mean we need to create multiple clients?
on OSX the following code consistently fails, What am I missing?

client.setID(1);
    client.readInputRegisters(5, 2, function(err, data) {
        if (err){
            console.log(err);            
        } else {
            console.log(data);
            var int32 = data.buffer.readFloatBE();
            console.log(int32);
        }

    });
    client.setID(2);
    client.readInputRegisters(6, 2, function(err, data) {            
            if (err){
                console.log(err);            
            } else {
                console.log(data);
                var int32 = data.buffer.readFloatBE();
                console.log(int32);
            }

        });

with "Error: Unexpected data error, expected 2 got 1"

yaacov, 2017-03-06:
> `var int32 = data.buffer.readFloatBE();`

do you mean `var float64 = data.buffer.readFloatBE();`

also you need to read 4 registers per float `... client.readInputRegisters(6, 4).then( ...`

see:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/buffer.js

yaacov, 2017-03-08:
@inpras did you find what was the problem ? if it still doesn't work we will need to open an issue for that.