yaacov/node-modbus-serial #46: Modbus request in response

biancode, 2017-01-22:
A [feature request](https://github.com/biancode/node-red-contrib-modbus/issues/3) needs an enhancement of the response to have data, response and request buffer, please

response {
 request: Buffer with CRC
 data : []
 buffer: Buffer
}

yaacov, 2017-01-25:
hi,

can you try to take:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/rtubufferedport.js
copy it call it debug port.

then add an emiter for `debug` messages , e.g.
when reciving data [line 72]
`this.emit('debug', this._buffer.slice(start, start + length));`
and when sending data [line 136]
`this.emit('debug', data);`

I am a little busy with other things, so if you will wait for me, it can take some time,
I think that this new debug port will do the trick :-)

yaacov, 2017-01-29:
fixed by #50 and #51

yaacov, 2017-01-30:
@connium @biancode hi,

Looking at the  [debug](https://www.npmjs.com/package/debug) module I only see the options to send to stdout/file.
Can we solve this issue using  [debug](https://www.npmjs.com/package/debug) module or some other standard way ?

biancode, 2017-01-30:
I'll try some more next week.

biancode, 2017-02-19:
for me yes - debug is perfect - but there is no debug output for now - right?  DEBUG=*  or DEBUG=modbus-serial doesn't work

connium, 2017-02-19:
You are right, there are no debug outputs in the code yet. Will add them in the upcoming week (I'm missing them sometimes as well :wink: ).

yaacov, 2017-02-19:
> I'm missing them sometimes as well 😉

👍

biancode, 2017-02-19:
node-red-contrib-modbus is working with debug npm from version 1.0.6+ and using the name 
`DEBUG=modbus-serial `
for modbus info and otherwise with the name 
`DEBUG=node_red_contrib_modbus`

closed 2017-03-04 by yaacov