yaacov/node-modbus-serial #56: Error: Cannot find module 'serialport'

abc-z, 2017-02-04:
I don't need `ModbusRTU` and I don't install `serialport` from `devDependencies`. When I include `modbus-serial` by `require('modbus-serial')` then my code failed with error `Cannot find module 'serialport'` because `index.js` try to use `rtubufferedport` who needs `serialports`.

Code below can fix that problem (imho it is not elegant way but it works)

// index.js:line 657
```
try {
  module.exports.RTUBufferedPort = require('./ports/rtubufferedport');
} catch(err) {}
```

yaacov, 2017-02-04:
Thanks!
Sorry you are right, i forgot why i didn't require it before ... , and just added it ...
can you make a pull request i can merge ?

yaacov, 2017-02-09:
fixed by https://github.com/yaacov/node-modbus-serial/pull/57

closed 2017-02-09 by yaacov