yaacov/node-modbus-serial #63: Modbus Serial Device Server

biancode, 2017-02-19:
Hi, 

now I've been testing a Modbus Serial Device Server (TCP to Serial NPort) and it has some errors.

a) if there is an internal restart/reset inside NPort (configuration), it raises no error about the lost TCP connection, but it crashes later see:
```
19 Feb 10:34:22 - TypeError: Cannot read property '_timeoutHandle' of undefined
    at TcpPort.<anonymous> (/.npm-global/lib/node_modules/node-red-contrib-modbus/node_modules/modbus-serial/index.js:211:43)
    at emitOne (events.js:96:13)
    at TcpPort.emit (events.js:188:7)
    at Socket.<anonymous> (/.npm-global/lib/node_modules/node-red-contrib-modbus/node_modules/modbus-serial/ports/tcpport.js:53:16)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:176:18)
    at Socket.Readable.push (_stream_readable.js:134:10)
    at TCP.onread (net.js:548:20) 
```

b) reading via TCP doesn't work, because of RTU buffered behavior - may it has to implemented in TCP buffered, too. It is possible to encapsulate RTU in TCP packages and in that case we got buffered behavior with split PDU data.

c) it works very well in writing data to NPort ;)

cya
Klaus

biancode, 2017-02-19:
see this [Modbus Link](http://www.rtaautomation.com/technologies/modbus-tcpip/)

yaacov, 2017-02-19:
Can you try to play with the telnetport to get what you need ?
https://github.com/yaacov/node-modbus-serial/blob/master/ports/telnetport.js

biancode, 2017-02-19:
telnet works a bit more - but - modbus-serial CRC error +0ms

yaacov, 2017-02-19:
> telnet works a bit more - but - modbus-serial CRC error +0ms

yes, telnet is for transmitting rtu packets over tcp (as if it was a serial line) , it's just a starting point ... if you can fix it to work, and then do a pull request, I can add a new port based on that ( ? tcp_buffered ? )

biancode, 2017-02-19:
I'll give it a try ;)

biancode, 2017-02-26:
test cases are done now and some lines of code - for documentation [image]

biancode, 2017-02-26:
new debug option start command:
DEBUG=modbus-serial node-red -v

comming code:
`var modbusSerialDebug = require('debug')('modbus-serial');`

`modbusSerialDebug('expected length of data is to small');`

biancode, 2017-03-03:
see first state [pull-Request](https://github.com/yaacov/node-modbus-serial/pull/66)

yaacov, 2017-03-04:
fixed by: https://github.com/yaacov/node-modbus-serial/pull/70

closed 2017-03-04 by yaacov