yaacov/node-modbus-serial #64: TCP reconnect doesn't work

biancode, 2017-02-19:
Hi,

since v4.6.1 (also 5.1.3) there is no error on lost TCP connection. How can I get that info from modbus-serial? Error is empty in that case. ( err{} )

before it raises the err.code === 'ECONNRESET' -  see with 4.5.1 it works

yaacov, 2017-02-19:
can you try to use promise on the command that trigger the error ? [ e.g. `.catch(function(err) { ... } )` ]

biancode, 2017-02-19:
I'm using these tow events:
    
```
node.client._port.on('close', node.onModbusClose)
node.client._port.on('error', node.onModbusError)
```

yaacov, 2017-02-19:
we already catch the `node.client._port.on('error' ...` here:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L70
maybe the promise `.catch(...` syntex will catch more errors ?

biancode, 2017-02-19:
also in catch empty -> modbus-serial {} +1s

yaacov, 2017-02-19:
> also in catch

hmm.... i will look into it more carefully ( probably next weekend )

biancode, 2017-02-19:
it would be great to have the Why it is not open as before - Connection Error or Reset or ... as TCP sends

biancode, 2017-02-19:
err.code I used before

yaacov, 2017-02-19:
we can not get an error from the `tcpport` for sending data to a close port because of this:
https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L602

but we can change the error code, will setting the error code to <SOMETHING> be ok ?

yaacov, 2017-02-19:
> it would be great to have the Why it is not open as before - Connection Error or Reset or ... as TCP sends

You should also get this from a `.catch` on the `connectTCP` see:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/simple.js#L8

biancode, 2017-02-19:
.catch works for me this way:  first I got Time out - plugged off the cable on working device - then later I got Port Not Open - net should tell you error 'ECONNRESET'

biancode, 2017-02-19:
possible codes - https://gist.github.com/gabrielfalcao/4216897

yaacov, 2017-02-19:
who is sending the errors: 
the `.catch ` is on the read/write command or the `.catch` on the `connectTCP` ?
the one on the  `connectTCP` should return the correct tcp error ... I will have to look into it ...

biancode, 2017-02-19:
.catch on read/write - connection is well done and then I unplug the device

biancode, 2017-03-05:
OK, now I got it to work very well.

`if(err.errno)`

then I can check the strings - errno is filled up in err object if we have a tcp error

`var networkErrors = ['ESOCKETTIMEDOUT', 'ETIMEDOUT', 'ECONNRESET', 'ECONNREFUSED'];`

closed 2017-03-05 by biancode

yaacov, 2017-03-05:
Can you give more info ... how to get this error ? where to put the `err.errno` check ? what port do you use ... ? do yo you know what fixed the problem ? ...

biancode, 2017-03-05:
There is no real fix - just more information in debugging - I just checked for ECONNRESET and the special case was ETIMEDOUT, if it doesn't reconnect
if the error comes from the TCP Socket, then I got the errno in error object - see new pull request examples - it was all fine with the promises and the use of .catch

yaacov, 2017-03-05:
👍 Nice :-)

srinivasyl, 2018-12-20:
I'm able to ping IP of Modbus PLC but, I'm not able to connect from the following script

```
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

client.connectTCP("192.168.9.2", { port: 502 });
client.setID(1);
 
setInterval(function() {
    client.readHoldingRegisters(3,1, function(err, data) {
        console.log(data);
    });
}, 1000);
```

the script throws the following error
" UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): Error: connect ECONNREFUSED 192.168.9.2:502"

help me to solve this error

yaacov, 2018-12-20:
Hi, @srinivasyl , can you try connect using some other tool ?
Maybe your device listen on some other port ? maybe your device does not answer to read holding registers, it can be many things ...

srinivasyl, 2018-12-26:
I've used Modbus pal simulator 
if I run Modbus pal in my system(IP: 127.0.0.1, PORT:502) the script was able to connect, read and write to registers but, If Modbus pal is running in the different machine it, then the script was unable to connect.
Is that network issue?

yaacov, 2018-12-26:
> Is that network issue?

try connect with the script from local machine ( like you did with the pal simulator ), if it also works, then it is probably network issue.