yaacov/node-modbus-serial #91: readCoils() only reading 8 bits

mstephenhunt, 2017-03-27:
Hey-- I think there may be an issue with the `readCoils()` method.

Running this code below, I'm always getting the same length of coils read no matter if I change the passed `count`:
```
client.readCoils(start, count)
  .then(function (received) {
    console.log('---> asked/received length: ', count, '/', received.data.length, received.data)
  })
```
`---> asked/received length:  1 / 8 [ false, false, false, false, false, false, false, false ]`
`---> asked/received length:  2 / 8 [ false, false, false, false, false, false, false, false ]`
`---> asked/received length:  20 / 8 [ false, false, false, false, false, false, false, false ]`

Am I misusing this method? I assumed that `data.length` should be the same as `count`. Thanks!

yaacov, 2017-03-27:
Thanks 👍

`received.data.length` should be: `ceil(length / 8) * 8` so:

> 1 => 8 is ok
> 2 => 8 is ok
> 20 => 8 is a bug ! should be 20

We take the 3rd byte [1] , this byte is the number of bytes in answer [2] and we take each bit as a boolean answer in an array, we do not stop when we reach `count` we do a full 8 bits loop [3].

I do not have a machine that answers readCoil available here:
a. Can you check what is the bug the return only 1 byte (8 bits) for `count = 20` ?
b. Can you make a pull request fixing it ?

[1] https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L42
[2] http://www.simplymodbus.ca/FC01.htm
[3] https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L48

mstephenhunt, 2017-03-28:
Sure thing!

I don't have our modbus machine right now, but I will tomorrow. I'll send over the results to you then. I'll see what I can do about a PR. Thanks for the confirmation!

yaacov, 2017-03-28:
np, thanks for the issue :-)

yaacov, 2017-04-01:
Checked v5.2.6 TCP and RTU, `readCoils()` return an answer with correct number of elements:
if N is the requested length answer will be of length  `ceil(N / 8) * 8`

Closed, fixed in current version

closed 2017-04-01 by yaacov