yaacov/node-modbus-serial #92: No reconnection client after disabling Modbus Server

joshuatreenl, 2017-03-29:
Hi,

Last week I post a message in the Node Red section of modbus. It seems that this issue is a problem in the node-modbus-serial. I have the following setup: One S7-1200 PLC and an IOT2000 device. On the IOT "Node Red" is installed and the node-red-modbus module. If I setup in the software a ModbusRead (function 3 read; read holding registers) then it seems to be working OK. Even if i disconnect (pull out the network connector) the connection is timedout and if connected the data is read again. BUT if I disable the software in the S7-1200 then the reconnection isn't working anymore. I get a CONNECTION REFUSED and then the Node Red will stop. I have contacted 'biancode' and he is pretty sure that the modbus-serial is maybe the problem. On his recommendation I tried to run the logger.js example. The results are: 

Connected
Receive: [ 8856, 0, 0, 17 ]
Receive: [ 8858, 0, 0, 17 ]
<disconnect the network physically>
Timed out
Timed out
Timed out
<connect the network physically)
Receive: [ 8870, 0, 0, 17 ]
Receive: [ 8871, 0, 0, 17 ]
Receive: [ 8872, 0, 0, 17 ]
Receive: [ 8879, 0, 0, 17 ]
^C <stop logger>

Started again the logger:
Connected
Receive: [ 9430, 0, 0, 17 ]
Receive: [ 9432, 0, 0, 17 ]
Receive: [ 9433, 0, 0, 17 ]
<disable the Modbus Server>
Port Not Open
Port Not Open
Port Not Open
<enable the Modbus Server>
Port Not Open
Port Not Open
Port Not Open
Port Not Open
Port Not Open
^C <stop logger>

The port won't be open again and will stay in this state. I did not get the REFUSED error...
Of course disabling the server is not likely to happen, but it can. My question is is there a solution to solve this problem. For example after a number of Port Not Open message, disconnect and try to connect again. Do this forever. 

Thanks in advance!

Rene

yaacov, 2017-03-29:
> For example after a number of Port Not Open message, disconnect and try to connect again. Do this forever.

Sounds great 👍 

I hope to have time this weekend, but can't promise :-)
Can you make a pull request that when finds a "Port not open" error [1],  tries to reconnect [2][3] ( for N times before giving up, or forever ? )

[1] https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L588
[2] https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L317
[3] https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L187

yaacov, 2017-03-29:
p.s. 
The reconnect can also be handled by the client when catching the "port not open" error, so maybe we can add an `errno` to the "port not open" error, to make it easier to catch ?

@biancode WDYT ?

biancode, 2017-03-29:
@yaacov that would be great to decide very clear what kind of problem we have on port not open

`'ECONNRESET', 'ENETRESET',
  'ECONNABORTED', 'ECONNREFUSED', 'ENETUNREACH', 'ENOTCONN',
  'ESHUTDOWN', 'EHOSTDOWN', 'ENETDOWN', 'EWOULDBLOCK', 'EAGAIN'` 

Posix socket should tell what is the problem in an ERRNO

yaacov, 2017-03-30:
@biancode @joshuatreenl it looks like the "close" signal can be called without an error [1] , can you try to see if this is the case here ? ( by adding a debug line [2] there )

[1] https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L69
[2] `modbusSerialDebug("port closed " + had_error); `

yaacov, 2017-03-30:
@biancode @joshuatreenl just added debugs [1] for the "error" and "close" signals in the tcp port, can you check with debug on [2][3] what happen when you turn off the controller ?

[1] https://github.com/yaacov/node-modbus-serial/commit/dbc6b781540766b3659d36890d8ae119fe50c91f
[2] https://github.com/visionmedia/debug
[3] in a unix shell, run `export DEBUG=modbus-serial` before running the logger.js program.

connium, 2017-03-30:
Hi guys,

I just want to give you a something to think about:
We at @meteocontrol use this library in a commercial product. This means we have to consider all kind of exceptions and handle them individually. So we check for disconnected errors, try to reconnect, do some error handling (depending on what went wrong and whether this kind of error occurred before) and so on.

If the modbus library, that is meant to handle a protocol, starts to deal with the connection, we'll get into trouble. IMO the handling of connection errors should be done outside in a client (I'm pretty sure the expectations of the actions in case of an error will differ for different users).

I'm fine with adding the errno to the exception but I strongly vote against reconnecting inside the `modbus-serial`.

yaacov, 2017-03-30:
> IMO the handling of connection errors should be done outside in a client 

@biancode @connium I agree 👍 
I do not understand how we ended up with a close port without some error being emitted ? @joshuatreenl can you use the code with the extra debug (in master) and run the logger.js example with debug on, we may understand why no error was sent, and solve this in a more precise way ?

joshuatreenl, 2017-03-30:
Sorry for the late answer. First of all thanks to the many usefull contributions by anyone. Let me explain the situation first. 

I have: 
S7-1200 type PLC and a IOT2040. 
On the S7-1200 I installed the Modbus TCP server, library from Siemens. I am able to disconnect the client in an odd manner. 
On the IOT2040 I installed Node Red with the packages S7 and Modbus (biancode). The S7 works fine but I would love to install a more general solution via Modbus function 3 (Read Holding Registers).  
The problem I got is that the ModbusRead function is not reconnecting after the odd disconnect of the server on the PLC. You could read the whole story here: https://github.com/biancode/node-red-contrib-modbus/issues/6
I am not the first contributer but somewhere halfway you see the discussion started with Biancode. He helps me a lot (many thanks!) and Biancode come to the conclusion that somewhere in the serial driver there is a disconnection and no reconnection. AND he makes use of the serial module to build the Node Red Modbus module...

I am:
New to IOT, Node Red and node.js... So my experience is not hmmm very high... sorry

Things to remember: 
The IOT is pretty new and image 2.1.2 is available. In that version node v4.2.4 is installed. I've tried the new version 7.7.4 because some packages in Node Red are not compiling without warning. The most packages want to have node >4.7.0. Within a couple of weeks a new image v2.1.3 should be available, but in that version 6.9.2 will be installed. Not the newest one...
Another point is the errorhandling and recovering within Node Red. This is pretty hard in my opinion. Therefor we have to decide if we want to use Node Red in the future. 

I agree with @connium , be carefull to change the code in that way that my problem is solved (I hope) but others are in trouble if this solution is implemented... 

I think, I agree with @yaacov , we should pinpoint where the problem occurs and then we have to decide what to do next (or not).

Before I test the version (with the extra lines in the code) I have to ask how I can recompile the code with the extra lines. I tried to install the package but there the tcpport.js is original. So I want to recompile without installing. Another point is that I hope that v4.2.4 is sufficient to recompile everyhting. If not I have to wait until the new image v2.1.3 is released.

Thanks in advance.

Rene

yaacov, 2017-03-31:
@joshuatreenl v5.2.4 is on npm
to install `npm install modbus-serial`
to run with debug on: https://github.com/visionmedia/debug

biancode, 2017-03-31:
@joshuatreenl node-red-contrib-modbus@1.0.9-theta now uses 5.2.4 of modbus-serial

joshuatreenl, 2017-03-31:
Ok fresh install of modbus-serial 

FIRST ATTEMPT: 

root@iot2000:/usr/lib/node_modules/modbus-serial/examples# DEBUG=modbus-serial node logger.js
  modbus-serial TCP port: signal connect +0ms
  modbus-serial { action: 'send tcp port',
  modbus-serial   data: <Buffer 01 03 00 05 00 04 54 08>,
  modbus-serial   buffer: <Buffer 00 00 00 00 00 06 01 03 00 05 00 04> } +318ms
  modbus-serial {"action":"send tcp port strings","data":{"type":"Buffer","data":[1,3,0,5,0,4,84,8]},"buffer":{"type":"Buffer","data":[0,0,0,0,0,6,1,3,0,5,0,4]}} +187ms
Connected
  modbus-serial { action: 'receive tcp port',
  modbus-serial   data: <Buffer 00 00 00 00 00 0b 01 03 08 00 00 bf 80 00 00 bf 80>,
  modbus-serial   buffer: <Buffer 01 03 08 00 00 bf 80 00 00 bf 80 ff 56> } +159ms
  modbus-serial {"action":"receive tcp port strings","data":{"type":"Buffer","data":[0,0,0,0,0,11,1,3,8,0,0,191,128,0,0,191,128]},"buffer":{"type":"Buffer","data":[1,3,8,0,0,191,128,0,0,191,128,255,86]}} +21ms
Receive: [ 0, 49024, 0, 49024 ]
  modbus-serial { action: 'send tcp port',
  modbus-serial   data: <Buffer 01 03 00 05 00 04 54 08>,
  modbus-serial   buffer: <Buffer 00 01 00 00 00 06 01 03 00 05 00 04> } +1s
  modbus-serial {"action":"send tcp port strings","data":{"type":"Buffer","data":[1,3,0,5,0,4,84,8]},"buffer":{"type":"Buffer","data":[0,1,0,0,0,6,1,3,0,5,0,4]}} +14ms
  modbus-serial { action: 'receive tcp port',
  modbus-serial   data: <Buffer 00 01 00 00 00 0b 01 03 08 00 00 bf 80 00 00 bf 80>,
  modbus-serial   buffer: <Buffer 01 03 08 00 00 bf 80 00 00 bf 80 ff 56> } +9ms
  modbus-serial {"action":"receive tcp port strings","data":{"type":"Buffer","data":[0,1,0,0,0,11,1,3,8,0,0,191,128,0,0,191,128]},"buffer":{"type":"Buffer","data":[1,3,8,0,0,191,128,0,0,191,128,255,86]}} +8ms
Receive: [ 0, 49024, 0, 49024 ]
  modbus-serial { action: 'send tcp port',
  modbus-serial   data: <Buffer 01 03 00 05 00 04 54 08>,
  modbus-serial   buffer: <Buffer 00 02 00 00 00 06 01 03 00 05 00 04> } +1s
  modbus-serial {"action":"send tcp port strings","data":{"type":"Buffer","data":[1,3,0,5,0,4,84,8]},"buffer":{"type":"Buffer","data":[0,2,0,0,0,6,1,3,0,5,0,4]}} +7ms
  modbus-serial { action: 'receive tcp port',
  modbus-serial   data: <Buffer 00 02 00 00 00 0b 01 03 08 00 00 bf 80 00 00 bf 80>,
  modbus-serial   buffer: <Buffer 01 03 08 00 00 bf 80 00 00 bf 80 ff 56> } +21ms
  modbus-serial {"action":"receive tcp port strings","data":{"type":"Buffer","data":[0,2,0,0,0,11,1,3,8,0,0,191,128,0,0,191,128]},"buffer":{"type":"Buffer","data":[1,3,8,0,0,191,128,0,0,191,128,255,86]}} +7ms
Receive: [ 0, 49024, 0, 49024 ]
  modbus-serial { action: 'send tcp port',
  modbus-serial   data: <Buffer 01 03 00 05 00 04 54 08>,
  modbus-serial   buffer: <Buffer 00 03 00 00 00 06 01 03 00 05 00 04> } +1s
  modbus-serial {"action":"send tcp port strings","data":{"type":"Buffer","data":[1,3,0,5,0,4,84,8]},"buffer":{"type":"Buffer","data":[0,3,0,0,0,6,1,3,0,5,0,4]}} +7ms
  modbus-serial { action: 'receive tcp port',
  modbus-serial   data: <Buffer 00 03 00 00 00 0b 01 03 08 00 00 bf 80 00 00 bf 80>,
  modbus-serial   buffer: <Buffer 01 03 08 00 00 bf 80 00 00 bf 80 ff 56> } +4ms
  modbus-serial {"action":"receive tcp port strings","data":{"type":"Buffer","data":[0,3,0,0,0,11,1,3,8,0,0,191,128,0,0,191,128]},"buffer":{"type":"Buffer","data":[1,3,8,0,0,191,128,0,0,191,128,255,86]}} +7ms
Receive: [ 0, 49024, 0, 49024 ]   
<RENE --- STOPPED SERVER 
  modbus-serial TCP port: signal errorError: read ECONNRESET +534ms
  modbus-serial TCP port: signal closetrue +5ms
Port Not Open
Port Not Open
Port Not Open
Port Not Open
Port Not Open
Port Not Open
Port Not Open
Port Not Open
Port Not Open
^C  <RENE --- STOPPED LOGGER.JS

SECOND ATTEMPT:
<RENE --- SERVER ENABLED AND STARTUP LOGGER.JS  
root@iot2000:/usr/lib/node_modules/modbus-serial/examples# DEBUG=modbus-serial node logger.js
  modbus-serial TCP port: signal connect +0ms
  modbus-serial { action: 'send tcp port',
  modbus-serial   data: <Buffer 01 03 00 05 00 04 54 08>,
  modbus-serial   buffer: <Buffer 00 00 00 00 00 06 01 03 00 05 00 04> } +332ms
  modbus-serial {"action":"send tcp port strings","data":{"type":"Buffer","data":[1,3,0,5,0,4,84,8]},"buffer":{"type":"Buffer","data":[0,0,0,0,0,6,1,3,0,5,0,4]}} +181ms
Connected
  modbus-serial TCP port: signal errorError: write EPIPE +181ms
  modbus-serial TCP port: signal closetrue +8ms
Timed out
Port Not Open
[... 17 lines cut ...]
  modbus-serial   buffer: <Buffer 01 03 08 00 00 bf 80 00 00 bf 80 ff 56> } +146ms
  modbus-serial {"action":"receive tcp port strings","data":{"type":"Buffer","data":[0,0,0,0,0,11,1,3,8,0,0,191,128,0,0,191,128]},"buffer":{"type":"Buffer","data":[1,3,8,0,0,191,128,0,0,191,128,255,86]}} +19ms
Receive: [ 0, 49024, 0, 49024 ]
  modbus-serial { action: 'send tcp port',
  modbus-serial   data: <Buffer 01 03 00 05 00 04 54 08>,
  modbus-serial   buffer: <Buffer 00 01 00 00 00 06 01 03 00 05 00 04> } +1s
  modbus-serial {"action":"send tcp port strings","data":{"type":"Buffer","data":[1,3,0,5,0,4,84,8]},"buffer":{"type":"Buffer","data":[0,1,0,0,0,6,1,3,0,5,0,4]}} +13ms
  modbus-serial { action: 'receive tcp port',
  modbus-serial   data: <Buffer 00 01 00 00 00 0b 01 03 08 00 00 bf 80 00 00 bf 80>,
  modbus-serial   buffer: <Buffer 01 03 08 00 00 bf 80 00 00 bf 80 ff 56> } +10ms
[... cut ...]
  modbus-serial { action: 'receive tcp port',
  modbus-serial   data: <Buffer 00 02 00 00 00 0b 01 03 08 00 00 bf 80 00 00 bf 80>,
  modbus-serial   buffer: <Buffer 01 03 08 00 00 bf 80 00 00 bf 80 ff 56> } +4ms
  modbus-serial {"action":"receive tcp port strings","data":{"type":"Buffer","data":[0,2,0,0,0,11,1,3,8,0,0,191,128,0,0,191,128]},"buffer":{"type":"Buffer","data":[1,3,8,0,0,191,128,0,0,191,128,255,86]}} +7ms
Receive: [ 0, 49024, 0, 49024 ]
^C <RENE --- STOPPED LOGGER.JS

I'll hope you can see what happens.
-Rene-

joshuatreenl, 2017-03-31:
Mentioned already in issue #93. we should close this issue. I am pretty sure that I have to build some error handling and reconnection within the Node Red application I have to develop. The driver is not the issue, how to follow up on errors is the issue. Sorry for everybody times to spent on this issue.

yaacov, 2017-03-31:
> Mentioned already in issue #93

@joshuatreenl #93 is a patch that may or may not fix this bug, if it does fix this, then we will merge it make a new version and close this issue ( and be very happy 🍰 ).

yaacov, 2017-03-31:
```
Receive: [ 0, 49024, 0, 49024 ]
<RENE --- STOPPED SERVER
modbus-serial TCP port: signal errorError: read ECONNRESET +534ms
modbus-serial TCP port: signal closetrue +5ms
Port Not Open
```

Than we have an error with errno ECONNRESET !! great

yaacov, 2017-03-31:
```
GREAT IT WORKS!
I will make a logfile via the DEBUG option.
```

Closing

yaacov, 2017-03-31:
> Bummer it does not work. I waited not long enough to make a connection again SORRY

OK reopen :-)

yaacov, 2017-03-31:
@biancode @joshuatreenl 
I just released v5.2.6, it fixes a bug that allow trying to write to a closed port, this bug occurred in v5.2.4 that i released earlier today for @joshuatreenl testing.

I think that it was also the version that started the odd behaviour of exiting the logger.js example when port is closed. if old nodejs versions could not handle writing to close port, it would explain this.

@joshuatreenl you can try the new version to see if now the logger does not exit when closing the server.
@biancode it's an important bug fix, you should update the version in node-red

yaacov, 2017-03-31:
link to current master:
https://github.com/yaacov/node-modbus-serial/archive/master.zip

yaacov, 2017-03-31:
looks like the problem was an old version of node, with node version ^6.9.0 it works.

closing.

closed 2017-03-31 by yaacov

joshuatreenl, 2017-04-19:
Hello @biancode and @yaacov, 
Today image v2.1.3 is released with node version 6.9.2. Although I've tested it for a short time:
ModbusRead is working OK with disconnect/connecting on the S7-PLC (E-CONNRESET). 
The problem with alloc buffer in ModbusServer is also solved.  

Thanks!

yaacov, 2017-04-19:
@joshuatreenl Thanks for the update !