yaacov/node-modbus-serial #95: Getting Crc Error in RTU

dedhdude, 2017-04-01:
Ascii is working completely fine but RTU is not working .
When I tried to communicate with two different device  in one im getting CRC error & in other im getting values but that also not true.
[image]

My Code

var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();
 
// open connection to a serial port 
client.connectRTU("/dev/ttyAMA0", {baudrate: 9600, dataBits: 8, parity: 'even', stopBits: 1}, read);
 
function read() {
	client.setID(1);
    // read the 2 registers starting at address 5 
    // on device number 1. 
    client.readHoldingRegisters(5, 2)
        .then(console.log);
}

And The one which im using in ardunio. This code is working fine.
char pubdata1[2000];
String mn = "1";
String vrv = "";  //6
String vyv = "";  //7
String vbv = "";  //8
String freq = "";  //9
String pd = "";  //
String cavg = "";  //11
String cr = ""; //12
String cy = ""; //13
String cb = "";  //14
String pfavg = ""; //
String w = "";
String wh = "";
String v1thd = "";
String v2thd = "";
String v3thd = "";
String a1thd = "";
String a2thd = "";
String a3thd = "";
String pfr = "";  //16
String pfy= "" ;  //17
String pfb = "";  //18
String fae = "";  // 
String ap = "";  //
String meterdata = "";
static union
{
    unsigned long a;
    byte b[4];
    float f;
    
}vr,vy,vb,ar,ay,ab,pfa,f;

unsigned int crcis(byte* buf,int len, int chk)
{
unsigned int crc = 0xffff;
unsigned int pos = 0;
byte crc0,crc1;
for(pos=0;pos<len;pos++)
{
  crc^=(unsigned int)buf[pos];
  for(int i=8;i!=0;i--)
  {
    if((crc & 0x0001)!=0)
    {
      crc>>=1;
      crc ^=0xA001; 
    }
    else
    crc >>=1;
  }
}

if (chk == 0)
{
  buf[pos] = crc & 0x00ff;
  buf[pos+1] = crc>>8;
  return 1;
}
else
{
    if (buf[pos] == (crc & 0x00ff))
    {
      if (buf[pos+1] == ((crc>>8) & (0x00FF)))
      return 1;
    }
    else
    {
      return 0;
    }
    
}
}

void readregister(unsigned int deviceid, unsigned int functioncode, unsigned int address)
{
  byte rxbuf[]={0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0};
  address = address - 1 ; 
  byte data[] = {deviceid, functioncode, highByte(address),lowByte(address),0x00,0x02,0x00,0x00};
  crcis(data,6,0);
  Serial.flush();
  for(int i=0;i<8;i++)
  {
    Serial.write(data[i]);
  }
  delay(250);
  while(Serial.available()>0)
  {
    for(int v=0; v<=8;v++)
    {
      rxbuf[v]=Serial.read();
     
    }
  }
  Serial.flush();
  
  vr.b[3]=rxbuf[5];
  vr.b[2]=rxbuf[6];
  vr.b[1]=rxbuf[3];
  vr.b[0]=rxbuf[4];
}
void setup_wifi()
  {
    //Serial.println();
    //Serial.print("Connecting to Network...");
    //Serial.println(ssid);
   // WiFi.begin(ssid,password);

    while(WiFi.status() != WL_CONNECTED)
    {
      delay(500);
     // Serial.print(".");
    }
    //Serial.println("");
    //Serial.println("WiFi connected");
    //Serial.println("IP address: ");
   // Serial.println(WiFi.localIP());
  }

} 
void callback(char* topic, byte* payload, unsigned int length) {
 // Serial.print("Message arrived [");
  //Serial.print(topic);
 // Serial.print("] ");
  for (int i = 0; i < length; i++) {
   // Serial.print((char)payload[i]);
  }
 // Serial.println();

  // Switch on the LED if an 1 was received as first character
  if ((char)payload[0] == '1') {
    DO1state = 1;   // Turn the LED on (Note that LOW is the voltage level
    // but actually the LED is on; this is because
    // it is acive low on the ESP-01)
  } else {
    DO1state = 0;// Turn the LED off by making the voltage HIGH
  }

}

void loop () {
    reconnect();  
    client.loop();
    DateTime nownow = rtc.now();
    
    for(int k=1;k<=1;k++)
   {
      DI1state = digitalRead(DI1);
      DI2state = digitalRead(DI2);
      DI3state = digitalRead(DI3);
      digitalWrite(DO1, DO1state);
      delay(10);
     /* readregister(1,3,3961);  //Wh 
      wh = String(vr.f);
      delay(10);
      Serial.flush();
      readregister(1,3,3959);  //forward apparent energy kvah
      fae = String(vr.f);
      delay(10);*/
    
      readregister(k,3,143);  //vr
      vrv = String(vr.f);
      delay(10);
      readregister(k,3,145);  //vy
      vyv = String(vr.f);
      delay(10);
      readregister(k,3,147);  //vb
      vbv = String(vr.f);
      delay(10);
      readregister(k,3,151);  //cr
      cr = String(vr.f);
      delay(10);
      readregister(k,3,153);  //cy
      cy = String(vr.f);
      delay(10);
      readregister(k,3,155);  //cb
      cb = String(vr.f);
      delay(10);
      readregister(k,3,157);  //freq
      freq = String(vr.f);
      delay(10);
      readregister(k,3,117);  //pfavg
      pfavg = String(vr.f);
      delay(10);
      readregister(k,3,101);  //Watt total
      w = String(vr.f);
      delay(10);
//      readregister(k,3,3861);  //V1%
//      v1thd = String(vr.f);
//      delay(10);
//      readregister(k,3,3863);  //v2%
//      v2thd = String(vr.f);
//      delay(10);
//      readregister(k,3,3865);  //V3%
//      v3thd = String(vr.f);
//      delay(10);
//      readregister(k,3,3867);  //A1%
//      a1thd = String(vr.f);
//      delay(10);
//      readregister(k,3,3869);  //A2%
//      a2thd = String(vr.f);
//      delay(10);
//      readregister(k,3,3871);  //A3%
//      a3thd = String(vr.f);
      delay(10);
      Serial.flush();
      mn = k;
        }
}

dedhdude, 2017-04-01:
Even Tried RTUBuffer

yaacov, 2017-04-01:
Try the `connectRTUBuffered`

```
client.connectRTU("/dev/ttyAMA0", {baudrate: 9600, dataBits: 8, parity: 'even', stopBits: 1})
  .then(read)
  .catch(console.log)
```
will become:
```
client.connectRTUBuffered("/dev/ttyAMA0", {baudrate: 9600, dataBits: 8, parity: 'even', stopBits: 1})
  .then(read)
  .catch(console.log)
```

yaacov, 2017-04-01:
can you check using the pymodbus:

a. install ( if you have python and pip )
```
$ sudo pip install pymodbus
...
```
b. run python
```
$ python
Python 2.7.13 (default, Jan 12 2017, 17:59:37) 
[GCC 6.3.1 20161221 (Red Hat 6.3.1-1)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from pymodbus.client.sync import ModbusSerialClient as ModbusClient
>>> client = ModbusClient(method='rtu', port='/dev/ttyAMA0', timeout=1)
>>> client.connect()
True
>>> client.read_holding_registers(5,2).registers
[234, 234]
>>> 
```

yaacov, 2017-04-01:
Checked against ArduinoESP running TCP and RTU modbus, no crc errors.

Arduino code:
https://github.com/yaacov/ArduinoModbusESP

Test used as client:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/simple.js

Fixed in v5.2.6 , closing.

closed 2017-04-01 by yaacov