yaacov/node-modbus-serial #96: Unable to close TCP connection

sujeethk, 2017-04-07:
`client.close();` for a TCP connection does not close the connection. The socket stays open on the slave and master. Is there a different method to close TCP connection?

yaacov, 2017-04-07:
Hi, currently the `client.close()` [1] method, only calls  `socket.end()` [2]

I do not want to change that behaviour, but we can add a `destroy` call if necessary, @connium WDYT ?

@sujeethk a workaround that may solve your problem:
calling `.destroy()` [3] directly on the tcp socket:
`client._client.destroy();`

[1] https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L97
[2] https://nodejs.org/api/net.html#net_socket_end_data_encoding
[3] https://nodejs.org/api/net.html#net_socket_destroy_exception

connium, 2017-04-08:
IMO the current behavior of `client.close()` is fine (see the link @yaacov referred as [3]).

If there is need for `socket.destroy()` I would place it in a separate method as @yaacov proposed. Users of this library should not be forced to use internal stuff (`client._client` should never be used outside the library).

yaacov, 2017-04-08:
Hi all, If the workaround described in https://github.com/yaacov/node-modbus-serial/issues/96#issuecomment-292584257 works, I will add a `.destory` function to the client as @connium suggested.

But we need to know that this was the problem, and that calling `client._client.destroy();` fixes it. @sujeethk can you try it, and comment here if this solve/not solve the problem ?

yaacov, 2017-04-15:
@sujeethk ping ^^

sujeethk, 2017-04-15:
Apologies for not getting back earlier. When I log `client._client` it does not have a destroy method. 

And I assume there is a wait timeout for the socket to close before being destroyed which is why the socket doesn't close immediately after calling close.

yaacov, 2017-04-15:
a. np :-)
b. `client._client` should be a socket ( it has a `destroy` method ) 
  1. can you write here the relevant code you are using ?  
  2. where/when socket is still connected, 
  3. where/when you see that `client._client` is not a socket anymore ( does not have a `destroy` method )

yaacov, 2017-04-19:
@sujeethk I didn't understand, is this working now ? do we need to close this issue ( it is solve already ) , or we need to fix it , how ?

sujeethk, 2017-04-19:
I think we can close this issue since it is working as designed. When we call close on the socket, it flags it for closing but waits until the default timeout to make sure there are no pending transmission from the other client. Once the timeout is done it closes the socket.

yaacov, 2017-04-19:
Thanks 👍  

closing

closed 2017-04-19 by yaacov