yaacov/node-modbus-serial #97: Wrte Only - Broad cast

inpras, 2017-04-25:
Is there any way to configure RTU Buffered to have write only request (as in broadcast request)?

yaacov, 2017-04-25:
What happen when you try, it should just work ?

inpras, 2017-04-25:
This is the code I have

```
client.writeRegister(currentParameter.mbAddress, currentParameter.value, function(err, data) {
                    if (!err){
                    } else {
                    }
                });
```

the call for writeRegister stays there forever, never comes out! I think it is expecting reply from slave device and for broadcast messages, we don't get reply!

yaacov, 2017-04-25:
what happen when you do not try to catch the error ?
e.g. `client.writeRegister(currentParameter.mbAddress, currentParameter.value)`

inpras, 2017-04-25:
No change with that also. There is a promise that keeps waiting if the callback is not supplied.

yaacov, 2017-04-25:
> There is a promise that keeps waiting if the callback is not supplied.

Yes, but a promise does not block, you can continue to use the port for other requests and ignore the waiting promise.

inpras, 2017-04-25:
OK, let me give a try to see what I was missing earlier, thank you for your response. Will get back to you with my results.

yaacov, 2017-04-25:
p.s. 
low level commands, can be called without a callback, e.g.
`client.writeFC6(currentParameter.mbAddress, currentParameter.value)`

inpras, 2017-04-25:
yes, that makes more informed of me. Thank you

yaacov, 2017-04-25:
np :-)

If this works, you can close the issue, if not, comment here.

inpras, 2017-04-27:
Thank you this would fit my need. I am closing this now.

closed 2017-04-27 by inpras