yaacov/node-modbus-serial #98: "Type error: Cannot read property '_timeouthandle' of undefined."

biancode, 2017-04-26:
[node-red-contrib-modbus issue](https://github.com/biancode/node-red-contrib-modbus/issues/18) reported by @matthiaswinkeler

yaacov, 2017-04-26:
@biancode hi, just added a check for  `Type error: Cannot read property '_timeouthandle' of undefined.`

Can you check it from git ?
I do not want to release a new version just for that, if this does not solve the issue :-)

matthiaswinkeler, 2017-04-26:
Hi biancode & yaacov 

Can I help you guys with more information? 

Thanks for your help!

yaacov, 2017-04-26:
@matthiaswinkeler thanks, 
The test we need is to use the current git code (with the fix) in `node-red-contrib`, and see if this fix the problem.

@biancode can you help @matthiaswinkeler to make a node-red module that use the git code instead of the current version (npm) ?

matthiaswinkeler, 2017-04-26:
Ok let me do some research :)

yaacov, 2017-04-26:
> Ok let me do some research :)

👍 I do not know the node-red modulae, but if you have questions, just ask ... :-)

matthiaswinkeler, 2017-04-26:
i tried "sudo npm intstall https://github.com/yaacov/node-mobdus-serial.git"

But the problem persists, I do think I didnt install the update correctly :)

yaacov, 2017-04-26:
@matthiaswinkeler their is no update ... it is just in the git.

I will make a new version , and then @biancode will make a vew version ... and then you could simply check ...

I am working on a release, wait ...

yaacov, 2017-04-26:
ok, new version 5.2.7

@biancode @matthiaswinkeler  you can now make the node-red module using the new version from npm.

https://www.npmjs.com/package/modbus-serial

biancode, 2017-04-26:
@yaacov _port.on("data", function(data) should be left if there is no transaction, because it makes nothing without a transaction, right?

biancode, 2017-04-26:
Maybe there is a bigger "problem" - why can I have on data events without a transactionId on reading? It will be set on request.

yaacov, 2017-04-26:
> _port.on("data", function(data) should be left if there is no transaction, because it makes nothing good without a transaction, right?

Yes 👍  i will fix that, thanks.

> Maybe there is a bigger "problem" 

`_transactionId` is something port get from outside, theoretically the controller can give us **any** number here [1] , so we must protect `_transactionId` as a rule.

[1] https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L55

yaacov, 2017-04-26:
p.s.
@matthiaswinkeler did you test with the new version ? did it solve the initial problem ?

matthiaswinkeler, 2017-04-26:
Hi,

I tested the new updates, with my old flow code 

but the error persisted. I deleted all flows (emptied them out) and connection nodes (all my slaves) but still the error was there

It always restarts around 60 second after server is running message. So actually the first 30 polls run fine.

biancode, 2017-04-26:
Yes, it doesn't work, if there is no transaction (of undefined means the object transaction).

biancode, 2017-04-26:
@matthiaswinkeler can you check if the problem is also in 

`npm install node-red-contrib-modbus@1.0.8`

matthiaswinkeler, 2017-04-26:
Hi,

Thanks again for helping out. I will be back with an answer on this tomorrow morning.

matthiaswinkeler, 2017-04-26:
@biancode  

after installin 1.0.8 and importing the flow code I deployed and so far everything works great! I will let it run overnight and check the log in the morning .

But it looks promising, thanks so much for the effort!!

yaacov, 2017-04-27:
> Yes, it doesn't work, if there is no transaction (of undefined means the object transaction).

@biancode ahhh... did you fix the `_port.on("data", function(data)` thing ? or does it work now because of a different change ?

yaacov, 2017-04-27:
> _port.on("data", function(data) should be left if there is no transaction, because it makes nothing good without a transaction, right?

@biancode 👍 Thanks, fixed in 5.2.8 ( in npm )

yaacov, 2017-04-27:
p.s. @biancode @matthiaswinkeler we also have a limit of 64 on the number of concurrent transactions [1] this also may be too low for some use cases

[1] https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L12

yaacov, 2017-04-27:
v5.2.9 is out :-) , it has a limit of 256 transactions.

biancode, 2017-04-27:
OK I'll Do the update and tests today

matthiaswinkeler, 2017-04-28:
Hi,

I get the error aigain. under contrib-modbus 1.08 and the newest modbus-serial.
Here are a few screenshots:

[image]

[image]

Do I have the wrong version installed? 

My node red code:
[{"id":"9f6e3f8b.04ef","type":"inject","z":"eadbb9d7.688fb8","name":"STATUS POLL RATE","topic":"","payload":"true","payloadType":"bool","repeat":"10","crontab":"","once":false,"x":160,"y":400,"wires":[["8d2f88ce.b76d68","33af198b.ace886","e7817311.8927b"]]},{"id":"5517693e.e5b828","type":"inject","z":"eadbb9d7.688fb8","name":"RPM POLL RATE","topic":"","payload":"true","payloadType":"bool","repeat":"1","crontab":"","once":false,"x":150,"y":440,"wires":[["25261ed2.d321d2","f20390d9.c0669","547c0135.bb189"]]},{"id":"8d2f88ce.b76d68","type":"modbus-getter","z":"eadbb9d7.688fb8","name":"STAT","showStatusActivities":true,"unitid":"","dataType":"HoldingRegister","adr":"531","quantity":"54","server":"426ed311.a07e5c","x":210,"y":500,"wires":[["ddc62d46.efd34"],[],[]]},{"id":"25261ed2.d321d2","type":"modbus-getter","z":"eadbb9d7.688fb8","name":"RPM","showStatusActivities":true,"unitid":"","dataType":"HoldingRegister","adr":"97","quantity":"2","server":"426ed311.a07e5c","x":230,"y":560,"wires":[["a85d9240.73b2c"],[],[]]},{"id":"33af198b.ace886","type":"modbus-getter","z":"eadbb9d7.688fb8","name":"STAT","showStatusActivities":true,"unitid":"","dataType":"HoldingRegister","adr":"531","quantity":"54","server":"6032b129.0d9e5","x":230,"y":640,"wires":[["2041e77c.9187a8"],[],[]]},{"id":"f20390d9.c0669","type":"modbus-getter","z":"eadbb9d7.688fb8","name":"RPM","showStatusActivities":true,"unitid":"","dataType":"HoldingRegister","adr":"97","quantity":"2","server":"6032b129.0d9e5","x":230,"y":700,"wires":[["6e02ebb5.1c10c4"],[],[]]},{"id":"e7817311.8927b","type":"modbus-getter","z":"eadbb9d7.688fb8","name":"STAT","showStatusActivities":true,"unitid":"","dataType":"HoldingRegister","adr":"531","quantity":"54","server":"edfda330.026ea","x":230,"y":780,"wires":[["218acb0c.040144"],[],[]]},{"id":"547c0135.bb189","type":"modbus-getter","z":"eadbb9d7.688fb8","name":"RPM","showStatusActivities":true,"unitid":"","dataType":"HoldingRegister","adr":"97","quantity":"2","server":"edfda330.026ea","x":230,"y":840,"wires":[["f7a65007.792f4"],[],[]]},{"id":"426ed311.a07e5c","type":"modbus-client","z":"","name":"HN","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"10.10.2.3","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"50","clientTimeout":"1000","reconnectTimeout":"2000"},{"id":"6032b129.0d9e5","type":"modbus-client","z":"","name":"HM","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"10.10.2.4","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"50","clientTimeout":"1000","reconnectTimeout":"2000"},{"id":"edfda330.026ea","type":"modbus-client","z":"","name":"HZ","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"10.10.2.5","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"50","clientTimeout":"1000","reconnectTimeout":"2000"}]

Thanks a lot again!

yaacov, 2017-04-28:
> under contrib-modbus 1.08

@matthiaswinkeler @biancode I think the version with the fix [1] is 1.0.11

[1] https://github.com/biancode/node-red-contrib-modbus/blob/master/package.json#L3

biancode, 2017-04-28:
/.npm-global/lib
└── node-red-contrib-modbus@1.0.11

biancode, 2017-04-28:
@matthiaswinkeler here a little optimised version of your last flow:

[{"id":"cda5d12.d2b7c3","type":"tab","label":"Flow 1"},{"id":"92fb9292.d61aa","type":"modbus-flex-getter","z":"cda5d12.d2b7c3","name":"HN","showStatusActivities":false,"showErrors":false,"server":"33e9c7c.6896438","x":650,"y":60,"wires":[[],[]]},{"id":"95a83eee.9a2e3","type":"function","z":"cda5d12.d2b7c3","name":"Status - request 54 from reg 531 ","func":"msg.payload = { value: msg.payload, 'fc': 3, 'unitid': 1, 'address': 531 , 'quantity': 54 } \nreturn msg","outputs":1,"noerr":0,"x":410,"y":120,"wires":[["92fb9292.d61aa","beb61488.944048","1097ad98.df31b2"]]},{"id":"53370b1f.520aa4","type":"inject","z":"cda5d12.d2b7c3","name":"STATUS POLL RATE","topic":"","payload":"true","payloadType":"bool","repeat":"10","crontab":"","once":false,"x":140,"y":120,"wires":[["95a83eee.9a2e3"]]},{"id":"beb61488.944048","type":"modbus-flex-getter","z":"cda5d12.d2b7c3","name":"HM","showStatusActivities":false,"showErrors":false,"server":"5ef4dfab.c2619","x":650,"y":120,"wires":[[],[]]},{"id":"1097ad98.df31b2","type":"modbus-flex-getter","z":"cda5d12.d2b7c3","name":"HZ","showStatusActivities":false,"showErrors":false,"server":"8016ac4e.7f5d7","x":650,"y":180,"wires":[[],[]]},{"id":"bac65ffc.62fd","type":"modbus-flex-getter","z":"cda5d12.d2b7c3","name":"HN","showStatusActivities":false,"showErrors":false,"server":"33e9c7c.6896438","x":650,"y":260,"wires":[[],[]]},{"id":"e6ea18b6.139d08","type":"function","z":"cda5d12.d2b7c3","name":"RPM - request 2 from reg 97 ","func":"msg.payload = { value: msg.payload, 'fc': 3, 'unitid': 1, 'address': 531 , 'quantity': 54 } \nreturn msg","outputs":1,"noerr":0,"x":400,"y":300,"wires":[["bac65ffc.62fd","adaa5c4.08ebba","eb44a17.4e9676"]]},{"id":"adaa5c4.08ebba","type":"modbus-flex-getter","z":"cda5d12.d2b7c3","name":"HM","showStatusActivities":false,"showErrors":false,"server":"5ef4dfab.c2619","x":650,"y":320,"wires":[[],[]]},{"id":"eb44a17.4e9676","type":"modbus-flex-getter","z":"cda5d12.d2b7c3","name":"HZ","showStatusActivities":false,"showErrors":false,"server":"8016ac4e.7f5d7","x":650,"y":380,"wires":[[],[]]},{"id":"c1e8a46f.9408d8","type":"inject","z":"cda5d12.d2b7c3","name":"RPM POLL RATE","topic":"","payload":"true","payloadType":"bool","repeat":"1","crontab":"","once":false,"x":150,"y":300,"wires":[["e6ea18b6.139d08"]]},{"id":"33e9c7c.6896438","type":"modbus-client","z":"cda5d12.d2b7c3","name":"HN","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"10.10.2.3","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"},{"id":"5ef4dfab.c2619","type":"modbus-client","z":"","name":"HM","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"10.10.2.4","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"},{"id":"8016ac4e.7f5d7","type":"modbus-client","z":"","name":"HZ","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"10.10.2.5","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

biancode, 2017-04-28:
[image]

matthiaswinkeler, 2017-04-28:
@biancode  

Looks nice!! Thanks I am going to run it right away ;)
I installed the 1.0.8 before because in a previous post this was recommended. 

I will in stall you new version and run the above code ;) 

Cheerss!

biancode, 2017-04-28:
[Video for you](https://www.youtube.com/watch?v=G4p-Af1olo8)

yaacov, 2017-05-02:
@biancode @matthiaswinkeler Hi, do we need to investigate more, or is this solved in 1.0.11 ?

matthiaswinkeler, 2017-05-02:
Hi,  thanks for your help on this. 

The issue is resolved now!

yaacov, 2017-05-02:
> The issue is resolved now!

closed 2017-05-02 by yaacov