yaacov/node-modbus-serial #104: Which method should I use to response Master？

HanSunyi, 2017-05-11:
Master send "03 03 00 00 00 06 C4 2A" to my "js slaver" per 500ms .

But which method should I use to response the master?

I know i should response the data from "0000" ,  length = 6   which response code is :
        03                    0C                 01 02 03 04 05 06 07 08 09 0A 0B 0C      XX XX
slave address      byte count        [_________________data________________]       CRC

but how to send this message to master?

yaacov, 2017-05-11:
Can you give a code example ?

HanSunyi, 2017-05-11:
@yaacov I find a variable named client._port._buffer
                when I am in debug mode,I can see my data exits in  client._port._buffer
                Should I directlly use client._port._buffer to control my code as below? 
                     something like 
```
reqBuffer = client._port._buffer;
    if (reqBuffer) {
        if(reqBuffer[0] == 3){  //
        //    03      03    0000     0006         C4 2A
        var startHight = reqBuffer[2].toString(16);   // 
        var startLow = reqBuffer[3].toString(16);   //
        var start16 = startHight + startLow; //

        var lenHight = reqBuffer[4].toString(16);   // 
        var lenLow = reqBuffer[5].toString(16);   // 
        var len16 = lenHight + lenLow;          //
        response(parseInt(start16,16),parseInt(len16,16)) // 
        }
    }
```

               Help me out please!

yaacov, 2017-05-11:
@HanSunyi hi, thanks

a. We currently only have tcp server [1][2] - a **server waits for MoabusTCP requests**, and responds to them.

b. We do not have a server for ModbusRTU :-(

c. You are using the client part of the library [3], this is the part that calls a server ( e.g. a controller ) , a **client send a request, and then waits for an answer**, see examples [4].

p.s
If you want to, you are very welcome to make a pull request adding a RTU server, copy the TCP one [2] and adjust it to listen to the serial port ...

[1] https://github.com/yaacov/node-modbus-serial/blob/master/examples/server.js
[2] https://github.com/yaacov/node-modbus-serial/tree/master/servers
[3] `var client = new ModbusRTU(); client.connectRTUBuffered("com6", { baudrate: 9600 });`
[4] https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger.js

HanSunyi, 2017-05-12:
@yaacov 
Thank you for your help and your patience.   : )
I just started using js,I still need to learn a lot of things.

yaacov, 2017-05-12:
np :-)

closed 2017-05-17 by yaacov