yaacov/node-modbus-serial #110: MODBUS RTU

rohitkumar5293, 2017-06-02:
Hi,
 i want to implement loopback test for HEXIN RS232 - RS485 converter via MODBUS. I am newbie to modbus implementation and i read you repository for modbus serial communication. My requirement is to do callback test rs232 to raspberry pi via modbus just simple read and write. Simply send some message to port ttylS0 and get same packet. Please advise me how can i implement this. Also unable to compile your code read/write sample which you given using node JS.

Please help me out.

rohit

yaacov, 2017-06-02:
> do callback test rs232 to raspberry pi via modbus just simple read and write

Do you have a controller listening on rs485/232 ?
Does it allow write to coils / registers ?

> Simply send some message to port ttylS0 and get same packet

https://github.com/yaacov/node-modbus-serial/blob/master/examples/debug.js

Change the port to `connectRTU` `ttyS0` and the `require` to  `require("modbus-serial");`

> Also unable to compile your code read/write sample which you given using node JS.

What do you mean ? please attach versions, code sample, error messages, debug output ...

yaacov, 2017-06-10:
@rohitkumar5293 any new information ?

closed 2017-06-12 by yaacov