yaacov/node-modbus-serial #114: .writeFC6(undefined, 100, 1) writing to all addresses

tekjanson, 2017-06-08:
So while using this I accidentally wrote to an undefined address and that caused everything on my network to receive that write. I checked with the below code and it happened again, I was wondering if this is supposed to be a feature of the library? it seems really dangerous, I looked and saw no documentation about this. 

// create an empty modbus client 
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();
 
// open connection to a serial port 
client.connectRTU("/dev/mydevice", { baudrate: 19200 }, write);
 
function write() {
    client.setID(1);
 
    // write the values 0, 0xffff to registers starting at address 5 
    // on device number 1. 
    client.writeFC6(undefined,100,1) //command to turn on light at address undefined turns on everything
        .then(read);
}
 
function read() {
    // read the 2 registers starting at address 5 
    // on device number 1. 
    client.readHoldingRegisters(5, 2)
        .then(console.log);
}

yaacov, 2017-06-08:
Nice, thanks ! 

We should protect the buffer [write functions](https://nodejs.org/api/buffer.html#buffer_buf_writeuint8_value_offset_noassert) before using them, for example here [FC6](https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L533) we should check if values are valid `int8` and `uint16` before continue to send them to the device.

Can you make a pull request that add this sanity checks before using the buffer [write functions](https://nodejs.org/api/buffer.html#buffer_buf_writeuint8_value_offset_noassert) ?

tekjanson, 2017-06-08:
I am really new to using node nor has I ever done that so I am not entirely sure how to do that actually

yaacov, 2017-06-08:
> I am not entirely sure how to do that actually

Before using the `address` variable we need to sanity check it ...
Something like this:

```
// sanity check
    if (typeof address === 'undefined' || typeof dataAddress === 'undefined' || typeof value === 'undefined') {
        if (next) next(new PortNotOpenError());
        return;
}
```

yaacov, 2017-07-09:
closed by: https://github.com/yaacov/node-modbus-serial/pull/121

closed 2017-07-09 by yaacov