yaacov/node-modbus-serial #115: PortNotOpenError

naveen26246, 2017-06-08:
Hi, 
I am trying to connect an energy meter(modbus) to my ubuntu 16.04 machine. However, when I run the following node app, I run into the error mentioned. 

**Error: 
PortNotOpenError {
  name: 'PortNotOpenError',
  message: 'Port Not Open',
  errno: 'ECONNREFUSED'}**

My code: 
// create an empty modbus client
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

// open connection to a serial port
client.connectRTUBuffered("/dev/ttyUSB0", { baudrate: 9600 });
client.setID(1);
client.readHoldingRegisters(0136, 2, function(err, data) {
   console.log(data.data);
});

Any leads to address the issue? 
Thanks in advance.

yaacov, 2017-06-08:
are you sure you have r/w rights to use `/dev/ttyUSB0` on your machine ?

naveen26246, 2017-06-08:
Yes I have added myself to the dialout group, rebooted and tried again. No change in the error.

yaacov, 2017-06-08:
Ahh....

try:
```
// create an empty modbus client
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

// open connection to a serial port
client.connectRTU("/dev/ttyUSB0", {baudrate: 9600}, run);

function run() {
    client.setID(1);

    client.readHoldingRegisters(136, 2, function(err, data) {
        console.log(data.data);
    });
}
```

naveen26246, 2017-06-08:
Hi, 

I tried the above code as suggested. There is no error. However, the data is not being printed.

yaacov, 2017-06-08:
>  There is no error.

Great 👍 

> However, the data is not being printed.

a. Are you shore baudrate, databits ...  are correct ?
b. Are you shore the device answer to this address ?

You can print the data:
`console.log(data);`
instead of:
`console.log(data.data);`

You can run in debug mode:
```
export DEBUG=modbus-serial
node program.js
```

yaacov, 2017-06-09:
p.s. 0136 is octal for 94 decimal not 136 decimal ...

naveen26246, 2017-06-09:
Am i supposed to use the octal value in the readHoldingRegisters function ?

yaacov, 2017-06-09:
> Am i supposed to use the octal value in the readHoldingRegisters function ?

in your code https://github.com/yaacov/node-modbus-serial/issues/115#issue-234522114 you write to address 94 ( 0136 ), in the code example I wrote here https://github.com/yaacov/node-modbus-serial/issues/115#issuecomment-307170804 we write to address 136 .

Assuming your device documentation say it answer to address 94 ( 0136 ), it may not answer to address 136 ...

yaacov, 2017-06-10:
@naveen26246 any new information ?

naveen26246, 2017-06-10:
@yaacov .. i might be having issues on the device end ... trying to get that fixed first. Will get back to you if I solve the issue ..

naveen26246, 2017-06-10:
@yaacov ... the issue was on the device.  Thanks for the support.

yaacov, 2017-06-10:
:-)

naveen26246, 2017-06-15:
Hi @yaacov , the module was working fine when I was testing my modbus device on my pc (ubuntu 16.04 , via uart-usb connector). However, when I try the same code on arm based device(ubuntu 16.04 again) , I get the following error. I am not sure if I need to edit serialport module and rebuild. Any help would be greatly apprecaited. 

  modbus-serial { action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 01 03 00 88 00 06 45 e2>,
  modbus-serial   unitid: 1,
  modbus-serial   functionCode: 3 } +0ms
  modbus-serial {"action":"send serial rtu buffered strings","data":{"type":"Buffer","data":[1,3,0,136,0,6,69,226]},"unitid":1,"functionCode":3} +118ms
  modbus-serial { action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 01 03 00 88 00 06 45 e2>,
  modbus-serial   unitid: 1,
  modbus-serial   functionCode: 3 } +1s
  modbus-serial {"action":"send serial rtu buffered strings","data":{"type":"Buffer","data":[1,3,0,136,0,6,69,226]},"unitid":1,"functionCode":3} +10ms
  modbus-serial { action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 01 03 00 88 00 06 45 e2> ..... 

My code: 

// create an empty modbus client
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

// open connection to a serial port
client.connectRTUBuffered("/dev/ttymxc2", { baudrate: 9600, parity:'even'});
client.setID(1);

// read the values of 10 registers starting at address 0
// on device number 1. and log the values to the console.
setInterval(function() {
    client.readHoldingRegisters(136, 6, function(err, data) {
        console.log(data.data);
    });
}, 1000);
Thanks,
Naveen

yaacov, 2017-06-15:
`/dev/ttymxc2` is connected to the same usb to [rs484/rs232/...] hardware you used on the PC ?

naveen26246, 2017-06-15:
No.. its directly connected to the rs 232 port on the gateway device.

yaacov, 2017-06-15:
a. your device speek modbus over rs232 ? 

b. you can connect the rs232 from you device to a PC and see what actually is sent ...

naveen26246, 2017-06-15:
a. My device speaks over modbus 485 (Sorry about the 232 confusion. Im a newbie here). 
b. The data is being read when i connect the modbus device to my pc. But not on the gateway device.

yaacov, 2017-06-15:
485 requires a direction pin , when you use a `usb -> 485` device this is done for you, you need to read the rs485 out documentation on your arm device.
If this direction pin is not set automatically by hardware, this library does not do this for you :-(

Scolioza, 2017-07-09:
Hello and thank you for this great piece of software. 
I want to implement a IOT project using Arduino and raspberry pi. For Arduino I used you library which, by the way, works great (callback idea is great). I tested the Arduino with QModbus software and everything worked flawless.
However the long term plan is to integrate everything on an express server. So I started to play with your modbus-serial node library on my computer planning to port the code to raspberry. 

Here is the code that I used:

```
// create an empty modbus client
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

// open connection to a serial port
client.connectRTU("COM7", {baudrate: 9600});
//client.setID(1);

// set a timout for requests default is null (no timeout)
client.setTimeout(1000);

// read 8 holding registers starting at register 10
// (function use the unit id 1)
client.writeFC3(1, 10, 8, function(err, data) {
      if (err) {
            console.log(err);
      } else {
            console.log(data);
          }
      });
```
But no matter what I do I get the same error:

```
PortNotOpenError {
  name: 'PortNotOpenError',
  message: 'Port Not Open',
  errno: 'ECONNREFUSED' }
```
Please give me some guidance.

Do I have to use the RS485 adapter? At this moment the connection is made via USB.

Thank you in advance.

Scolioza, 2017-07-09:
Just for testing purpose I did the following:

```
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();
var ev =require('events');
var em =new ev.EventEmitter();
// open connection to a serial port
client.connectRTU("COM7", {baudrate: 9600},onConn);
//client.setID(1);

// set a timout for requests default is null (no timeout)
client.setTimeout(10000);

function onConn(){
  console.log('connected');
  em.emit('connected');
}

// read 8 holding registers starting at register 10
// (function use the unit id 1)

em.on('connected',function(){
  console.log('here');
  client.writeFC3(1, 10, 4, function(err, data) {
        if (err) {
              console.log(err);
        } else {
              console.log(data);
            }
        });

});

```
The error from callback:

```
Error: Timed out
    at Timeout._onTimeout (D:\nodeJs\modbus\node_modules\modbus-serial\index.js:160:18)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)
```

yaacov, 2017-07-09:
@Scolioza Thanks, 

a. I do not see the 'connected' and 'here' strings in the output ... It looks like it never connected to the serial port "com7"
b. If you can solve it and post a pull request that fix it, it will help a lot (I do not have lots of time lately).

some notes I can think of that may help:
a. Try to connect to the serial port using https://github.com/EmergingTechnologyAdvisors/node-serialport and see if you can get more information about the connection.
b. Try the promise syntax, it will make things more clear (who is calling who when)
```
client.connectRTU("/dev/ttyUSB0", { baudrate: 115200 })
    .then(function() {
        console.log("Connected"); })
    .catch(function(e) {
        console.log(e.message); });
```

yaacov, 2017-07-09:
@Scolioza 
> Do I have to use the RS485 adapter? At this moment the connection is made via USB.

This is high level lib, it will work with both, it assume all low level stuff is taken care of by node-serialport.

Scolioza, 2017-07-09:
Thanks for your fast response.
I always get this response:

```
connected
here
Error: Timed out
    at Timeout._onTimeout (D:\nodeJs\modbus\node_modules\modbus-serial\index.js:160:18)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)

```
And with your snippet I still get Connected but when I want to write to register nothing happen.

yaacov, 2017-07-09:
> when I want to write to register nothing happen.

a. are you sure the unit id is 1 ?
b. are you sure baud, parity, stopbit ... are all currect ?
c. can you sniff what QModbus is sending and receiving:
   1. understand from that what are the unit-id, registers, function used ?
   2. compare to the debug log of node-modbus and see what is wrong ?

Scolioza, 2017-07-09:
The id is indeed 1. The settings are:
baudrate 9600
Databits 8
Stopbits 1
Parity none. 

QModbus is working as expected. I am able to read and write all the registers that are defined in arduino sketch. 
I will dig more this evening.

Scolioza, 2017-07-09:
I finnaly :) manage to make it work !!!

In my case autoOpen must be disabled.
So here is the code. Hope it will help somebody.

```
var SerialPort = require("serialport");
var serialPort = new SerialPort("COM7", {baudrate: 9600, autoOpen: false});

// create a modbus client using the serial port
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU(serialPort);

// open connection to a serial port
client.open();

var x=setInterval(function(){
  if(client.isOpen()){
    client.setID(1);
    client.writeFC3(1, 10, 8, function(err, data) {
      if (err) {
            console.log(err);
      } else {
            console.log(data);
      }
  })
}},1000);
```
Thanks again for your great job !

yaacov, 2017-07-09:
@Scolioza Thanks 👍 

Closing this, with your work around, mainly because I do not have time to fix this ...

p.s. can you try:

```
client.connectRTU("/dev/ttyUSB0", { baudrate: 9600, autoOpen: false })
    .then(function() {
        console.log("Connected"); })
    .catch(function(e) {
        console.log(e.message); });
```

closed 2017-07-09 by yaacov