yaacov/node-modbus-serial #116: Multiple errors when running modified Modbus server example

checor, 2017-06-12:
Hello,
I'm trying this library. I choose it because seems easier to use and it's pure js, compared to another libraries. Congrats on your work!

However, I feel I'm missing something too obvious since almost every time something someone connects and ask for data, the programs crashes.

I added some parameters that changes over time to the modbus server example, in order to simulate a PLC.

I uploaded the snippet here: https://gist.github.com/checor/6217b145a6601f6aff81da1cd6df092d

I added some variables that need to change over time. I don't know if it's interferes with the Modbus library, I'm a js amateur 😞 

Here are the error strings I encounter usually:

```
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: read ECONNRESET
    at exports._errnoException (util.js:1022:11)
    at TCP.onread (net.js:569:26)
```
Almost everytime a PLC disconnects/shuts down. From what I found about this error, I need to handle it myself, since it's a connection reset.

```
C:\modbus-sk\node_modules\modbus-serial\servers\servertcp.js:105
        responseBuffer: responseBuffer.toString("hex")
                                      ^

TypeError: Cannot read property 'toString' of undefined
    at _parseModbusBuffer (C:\modbus-sk\node_modules\modbus-serial\servers\servertcp.js:105:39)
    at Socket.<anonymous> (C:\modbus-sk\node_modules\modbus-serial\servers\servertcp.js:391:34)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:176:18)
    at Socket.Readable.push (_stream_readable.js:134:10)
    at TCP.onread (net.js:548:20)
```
When reading a coil. With this error I don't have a clue what is the cause.

Any assistance would be greatly appreciated.

xalepo4, 2017-06-12:
Hi, I'm using a server to simulate a modbus component and ECONNRESET happen sometimes. I don't know why, if someone can help. Thanks

yaacov, 2017-06-12:
@checor hi, thanks for the issue, I will look into it this weekend.

p.s.
Your snippet look ok, so it's probably something with the lib. You can run the programs ( simulator and reader ) with debug, maybe you will see something you do not see now:

```
export DEBUG=modbus-serial
node simulator.js
```

checor, 2017-06-12:
@yaacov Thanks for the fast reply! I will run the snippet with debugging enabled.
If I find anything, I will post the results.

uwburn, 2017-07-12:
Hello,
i'm using the Modbus Server to mock a controller where i should connect, and every time i restart the client (master) application the server crashes.

I think the server should handle the socket error event that happens when the client abruptly closes the connection. Alternative, an event might be exposed to be handled by user's code.

As a workaround i'm using:
`process.on('uncaughtException', function (err) {
    console.error(err.stack);
});`

Anyway, @yaacov big thumbs up for the library!

yaacov, 2017-07-12:
@uwburn Thanks

p.s.
If you can make a pull request with a fix, it will be very nice, I do not find the time to fix this ...

checor, 2017-07-12:
I will try to help with that too. I want to get my hands in Node.

uwburn, 2017-07-12:
@yaacov: i will try to get a grasp on it.

uwburn, 2017-07-12:
Submitted pull request #122.

yaacov, 2017-07-13:
Will merge once tests are green.

uwburn, 2017-07-13:
On windows the linter went crazy, so i just ignored it... my bad. I tested again on linux and fixed linter complaints. 

The bad side is that on Linux i could not get the ECONNRESET error to trigger on the server.
I'm spawning a child process that opens the connection to the server and then killing it: on Windows when the process is killed the socket is closed abruptly and ECONNRESET is triggered, but on Linux it doesn't happen.

I think i cannot implement a test for this, is that ok?

yaacov, 2017-07-13:
> I think i cannot implement a test for this, is that ok?

yes, thanks for the work ! I will try to make the test work and merge it when I do, you can try too :-).

uwburn, 2017-07-13:
Ok, i later on i will update the PR. I will leave the test case for ECONNRESET commented, so the test will not fail and you can see the idea behind it.

yaacov, 2017-07-15:
fixed by #122

closed 2017-07-15 by yaacov