yaacov/node-modbus-serial #117: connectTcpRTUBuffered always timeout

xalepo4, 2017-06-12:
Hello, 
I'm working with this library but when I use the server.js and then buffertcp.js it always give me timeout.
If I replace connectTcpRTUBuffered for connectTCP it works well, but I need the other function.
Any assistance would be greatly appreciated.

yaacov, 2017-06-12:
@xalepo4 hi, thanks for the issue, I will look into it this weekend.

If you can post example code that reproduce this, it will be easier to find the problem.

xalepo4, 2017-06-13:
This is the server I'm running:

`/* eslint-disable no-console, no-unused-vars, spaced-comment */

// create an empty modbus client
//var ModbusRTU = require("modbus-serial");
var ModbusRTU = require("../index");
var vector = {
    getInputRegister: function(addr) { return addr; },
    getHoldingRegister: function(addr) { return addr + 8000; },
    getCoil: function(addr) { return (addr % 2) === 0; },
    setRegister: function(addr, value) { console.log("set register", addr, value); return; },
    setCoil: function(addr, value) { console.log("set coil", addr, value); return; }
};

// set the server to answer for modbus requests
console.log("ModbusTCP listening on modbus://0.0.0.0:8502");
var serverTCP = new ModbusRTU.ServerTCP(vector, { host: "0.0.0.0", port: 8502, debug: true, unitID: 1 });`

And this is the client:

`/* eslint-disable no-console, spaced-comment */

// create an empty modbus client
//var ModbusRTU = require("modbus-serial");
var ModbusRTU = require("../index");
var client = new ModbusRTU();

// NPort Gateway 801D NetPort
client.connectTcpRTUBuffered("127.0.0.1", { port: 8502, removeCrc: false })
    .then(setClient)
    .then(function() {
        console.log("Connected");
    })
    .catch(function(e) {
        console.log(e.message);
    });

function setClient() {
    // set the client's unit id
    // set a timout for requests default is null (no timeout)
    client.setID(1);
    client.setTimeout(2000);

    // run program
    run();
}

function run() {
    // read the 4 registers starting at address 5
    client.readHoldingRegisters(5, 4)
        .then(function(d) {
            console.log("Receive:", d.data);
        })
        .catch(function(e) {
            console.log(e.message);
        })
        .then(close);
}

function close() {
    client.close();
}`

xalepo4, 2017-06-13:
This is the server I'm running:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/server.js
And this is the client:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/buffertcp.js

yaacov, 2017-06-13:
Hi, this tests run fine on my machine, do I miss something:

[image]

xalepo4, 2017-06-13:
I've try it again and this is the result:
Client.js
[image]
Server.js
[image]
And this is the output:
Client.js
[image]
Server.js
[image]

yaacov, 2017-06-13:
can you check the version you use ?

`npm view modbus-serial version`

xalepo4, 2017-06-13:
Version 5.3.1

yaacov, 2017-06-13:
can you try ?

`npm install modbus-serial@5.2.9`

xalepo4, 2017-06-13:
It works!!

yaacov, 2017-06-13:
Thanks for the Issue ! 
I will try to see what is wrong with the new version this weekend.

not closing this issue, until it is fixed in the new version.

p.s.
@checor we may have a bug in newer versions, can you try v5.2.9 ?

yaacov, 2017-07-15:
fixed by #123

closed 2017-07-15 by yaacov