yaacov/node-modbus-serial #118: Shouldn't serialport be an optional dependency?

vshymanskyy, 2017-06-14:
Also, Modbus RTU may be transported over virtually any streaming connection, like sockets or bluetooth (BLE) ,etc.

yaacov, 2017-06-14:
Yes, it was :-)

But we wanted to make it easy for users to just have one requirement,
See: https://github.com/yaacov/node-modbus-serial/issues/56

I'm inclining to remove this dependency ( we do not enforce it in the code ... [1] ) but I also want users to have easy life ...

[1] https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L656

vshymanskyy, 2017-06-14:
Cool. Thanks for explanation!
How about just being able to work with any Node.js stream: https://nodejs.org/api/stream.html
SerialPort already supports that and they improve it further in release 5+

yaacov, 2017-06-14:
> How about just being able to work with any Node.js stream:

sound great, can you make a pull request ?

vshymanskyy, 2017-06-14:
Probably - when I get to it ;)

yaacov, 2017-06-14:
> Probably - when I get to it ;)

have fun =^..^=

yaacov, 2017-07-09:
Closing, as I do not have time to look at it now.

closed 2017-07-09 by yaacov

emes, 2019-01-18:
Would you re-consider making serialport an optional dependency?  Docker images become muuuch larger when serialport needs to be built from source, which is shame when using only tcp.

yaacov, 2019-01-19:
@ems Thank you for this issue :+1: 

a - it's still optionals in the code: https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L738

b - I think the only problem is only here https://github.com/yaacov/node-modbus-serial/blob/master/package.json#L41

c- Can you think of a mechanism that will keep the default, **and** let us an option to build leaner containers ?
e.g. `package-no-serial-port.json`, `pckage.json.template` build with some Makefile ... ?

p.s. you can send a pull request with the dockerfile if you think it's nice :-)