yaacov/node-modbus-serial #119: TCP Server give differents responses each time

xalepo4, 2017-06-15:
Hi,
I'm using version 5.2.9 and I've found another extrange think.
If I execute the example TCP server and execute another script using the function readCoils for example it give me differents results everytime. If I execute readHoldingRegisters as in your tests it answer me ok, but only that function works correctly.
I attach my code and screenshots:

Server.js:
[image]

Client.js
[image]

Output with debug activated:
[image]

yaacov, 2017-06-15:
Thanks ! I will look into it

xalepo4, 2017-06-15:
I'm looking the code and testing and in servertcp.js when you create a buffer you use new Buffer(size) instead of new method Buffer.alloc(size). If you print the buffer before create with the old method it have random values.

xalepo4, 2017-06-15:
Tested changing new Buffer(size) to Buffer.alloc(size) and it works fine

yaacov, 2017-06-15:
Can you make a pull request ?

xalepo4, 2017-06-15:
Yes this afternoon I'll make.

yaacov, 2017-06-15:
Fixed in https://github.com/yaacov/node-modbus-serial/pull/120

Thanks @xalepo4

closed 2017-06-15 by yaacov