yaacov/node-modbus-serial #125: Modbus RTU (RPi 3 Model B to PC) Serial Sending but not Receiving

code19-io, 2017-07-19:
I have been testing my Raspberry Pi 3 Model B with my PC (Windows 10 64 Bit) using Serial communication to emulate Modbus slave and Master connection types. 
I have gone through multiple tutorials and i have been using a USB to Serial Converter (CH340) to communicate. 

Testings

- Connected GPIO15 and GPIO16, and communicated using ttyAMA0 port :  Works
- Connected my USB to Serial Converter to Raspberry Pi and a RS232 - TTL converter to GPIO15, GPIO16 and 3.3v : Works
- Connected my PC with USB to Serial Converter and Raspverry Pi with RS232 - TTL converter : Raspberry Writes to the Computer but not Reading (Partially Working)

Problem Scenario
Raspberry Pi (Works as a Modbus Master) connects to the PC (Works as a Modbus Slave). A Modbus Slave simulator is running on my PC with 
- baudRate: 9600
- Parity: None
- ByteSize: 8bits
- RTU
- StopBits: 1
- RTS 

And my Raspberry Pi has the following serial.py

```
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

client.connectRTU("/dev/ttyAMA0", { baudrate: 9600 }, write);

function write() {
    client.setID(1);
    client.writeRegisters(5, [0 , 0xffff])
        .then(read);
}

function read() {
    client.readHoldingRegisters(5, 2)
        .then(console.log);
}
```
The above function runs and changed the Modbus Simulator Input Register values but not Reading the Holding Registers ? 

Thank you in advance

yaacov, 2017-07-19:
hi, thanks for the issue

a. try to catch errors:
```
  .then(read)
  .catch(function(e) {
     console.log(e.message);  });
```

If you have errors, paste them here.

b. if you have another serial converter try to connect PC to PC and see if you can learn something new.

if it works / not works, describe what you see.

code19-io, 2017-07-19:
Hi @yaacov ,

Thanks for the quick reply.

I have tried your solution A.
The function it self is freezed and shows nothing.

Scenario

```
'use strict';

var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

client.connectRTU("/dev/ttyAMA0", { baudrate: 9600 }, write);

function write() {
    client.setID(1);
    console.log("Writing"); // Console Writing
    client.writeRegisters(5, [0 , 0xffff])
        .then(read).catch(function(err) { console.log(err.message); });
}

function read() {
    client.readHoldingRegisters(5, 2)
        .then(console.log);
}
```
When i run it from Raspberry pi, It prints out "Writing" and stays still, nothing happened.

Console Output
`Writing
`

B. I dont have another serial converter ... but i am pretty sure that this serial converter is working fine.

Thanks in Advance.

yaacov, 2017-07-19:
a. you can also try run with debug on:

```bash
export DEBUG=modbus-serial
# run you test program now
node test.js
```

b. add more `console.log` s , maybe you can understand somthing ?

yaacov, 2017-07-19:
do the simulator has debug data ?

code19-io, 2017-07-19:
I'll try to run with debug on,
If you dont mind can you walkthrough on how to debug (I am a Newbie)

What do you mean by Simulator has Debug Data ? 
The Simulator i am using is from http://www.modbustools.com/modbus_slave.html.

code19-io, 2017-07-19:
This is the output of DEBUG=modbus-serial node test.js
```
pi@raspberrypi:~/Desktop/code19_11-modbus-nodejs-fc1664a606ff $ DEBUG=modbus-serial node write.js
Writing
  modbus-serial { action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 01 10 00 05 00 02 04 00 15 00 16 a3 9a>,
  modbus-serial   unitid: 1,
  modbus-serial   functionCode: 16 } +0ms
  modbus-serial {"action":"send serial rtu buffered strings","data":{"type":"Buffer","data":[1,16,0,5,0,2,4,0,21,0,22,163,154]},"unitid":1,"functionCode":16} +18ms

```

yaacov, 2017-07-19:
> What do you mean by Simulator has Debug Data ?

a.
this:
http://www.modbustools.com/slave_traffic.html
should correspond to what you see in the test.js debug output.

b. From what I see, it looks like the message sent back from the PC never gets back to the Raspberry.

code19-io, 2017-07-19:
This is the output of Slave Traffic, 

[image]

> b. From what I see, it looks like the message sent back from the PC never gets back to the Raspberry.
Yes, I think so.

I have used [Hercules](http://www.hw-group.com/products/hercules/index_en.html) to send and receive serial communication. It Receives from the Raspberry Pi but when i started to send a message nothing happens.

I have inserted a 
`console.log("Reading")`

inside the Read function, but it appears to be not executed.

yaacov, 2017-07-19:
> It Receives from the Raspberry Pi but when i started to send a message nothing happens.

ok, it looks like this is the problem ...

code19-io, 2017-07-19:
Is it possible to sort it out?? Can you help me??

 Thanks in advance

yaacov, 2017-07-19:
> Is it possible to sort it out?? Can you help me??

It looks like a problem on the PC/Simulator side I have no way to help you with this ...

yaacov, 2017-07-19:
you should read the documentation of your USB to Serial Converter, maybe you are missing something in it's setting.

code19-io, 2017-07-19:
Thank you @yaacov for your guidance
I am closing this issue

closed 2017-07-19 by code19-io

yaacov, 2017-07-19:
thanks, you can comment here again if you find how to solve it.