yaacov/node-modbus-serial #126: Modbus TCP Write

IlhanSeven, 2017-07-23:
Thank you for sharing this. This has been very useful. I don't know if this functionality is implemented but Modbus TCP write gives:

{ name: '', message: 'Port Not Open', errno: 'ECONNREFUSED' }

Port is actually open because I am reading values via Modbus TCP. I wonder if it means 'Serial Port' when it says 'Port Not Open'.

yaacov, 2017-07-23:
Thanks for the issue, can you put here a sample program that show this ?

IlhanSeven, 2017-07-23:
Thanks for quick replay. The issue is timing as I found.
Below says 'Port Not Open'
...
client.connectTCP('192.168.1.68', { port: 5502 });
client.setID(1);
client.writeRegisters(5, [5, 6, 7]);
...

Below is ok.
...
client.connectTCP("192.168.1.68", { port: 5502 })
    .then(setClient);

function setClient() {
    client.setID(1);
    write();
}

function write() {
    client.writeRegisters(5, [5, 6, 7]);
}

IlhanSeven, 2017-07-23:
Yes, Thanks again.

closed 2017-07-23 by IlhanSeven

yaacov, 2017-07-23:
np :-)