yaacov/node-modbus-serial #129: Config Serial port parameters

q-bird, 2017-08-06:


q-bird, 2017-08-06:
Hi there,

I saw some examples in modbus-serial, when we use it with serial port, I just saw we can setup the baudrate like this: `client.connectRTU("/dev/ttyUSB0", { baudrate: 115200 })`

What about other parameters like `parity, stopbit, databit...` How to change them?

yaacov, 2017-08-06:
> What about other parameters like parity, stopbit, databit... How to change them?

all options available in node-serialport should work, e.g. `{ baudrate: 115200, parity: ..., stopbit: ..., ... }`

q-bird, 2017-08-06:
Hi
I saw in file `serialport-terminal.js`:
```
var openOptions = {
  baudRate: args.baud,
  dataBits: args.databits,
  parity: args.parity,
  stopBits: args.stopbits
};
```
So the keywords will be `baudRate, dataBits, parity, stopBits` right?

But when I setup the `client.connecRTU`, seems that the parameters not affect to the result
Meaning if I leave it as default, or change baudRate, dataBits... the result is also good (although the modbus slave I changed to the different config (different baudrate, databits...)

Is it auto-detect the config? and not depends on the parameters?

FYI: for example:
- in Nodejs: `client.connectRTU("/dev/ttyUSB0" )`
- in Modbus slave: any baudrate, any databits, any stopbits, any parity

Or even I set `client.connectRTU("/dev/ttyUSB0", { baudRate: 115200, dataBits: 8 })`
But it can also work with any parameters in modbus slave.... meaning not depend on the parameter in Nodejs, seems that it can detect the slave's parameter and re-config?

yaacov, 2017-08-06:
> But when I setup the client.connecRTU, seems that the parameters not affect to the result

👍 thanks I didn't notice that

> Nodejs, seems that it can detect the slave's parameter and re-config?

Not as far as I know ... but maybe \o/ 

in modbus-serial we just copy the options and send them to node-serial:
https://github.com/yaacov/node-modbus-serial/blob/master/apis/connection.js#L71

options we did not defined should default to:
https://github.com/EmergingTechnologyAdvisors/node-serialport#module_serialport--SerialPort..openOptions

p.s.
Thanks for the issue:
I do not have hardware to check this here, I will try to check it next weekend.

q-bird, 2017-08-06:
Thanks for your quick respond!

BTW, I also don't use hardware to test, I used a soft to create the virtual COM port to test with a Slave tool.
I will check with hardware to see later :D

yaacov, 2017-09-24:
Closing, this works for me :-)

p.s.
@q-bird feel free to re-open if this is not working for you on a real serial port ... we will need to see what is different in your set-up.

closed 2017-09-24 by yaacov