yaacov/node-modbus-serial #130: ServerTCP event

q-bird, 2017-08-08:
Hi,

```
var vector = {
    getInputRegister: function(addr, unitID) { return addr; },
    getHoldingRegister: function(addr, unitID) { return addr + 8000; },
    getCoil: function(addr, unitID) { return (addr % 2) === 0; },
    setRegister: function(addr, value, unitID) { console.log("set register", addr, value); return; },
    setCoil: function(addr, value, unitID) { console.log("set coil", addr, value); return; }
};
```
`var serverTCP = new ModbusRTU.ServerTCP(vector, { host: "0.0.0.0", port: 8502, debug: true });`

For example, the modbus client TCP sends a request to read 10 holdingRegs (addr from 0-9)
What I tested is the vector will be called 10 times continuously, each time for each addr (0, 1, ..., 9) and return the corresponding value.

*My issues are:*
- How to know when will the client modbus send a request? meaning how to get that event to control other functions? Because I want other functions to be implemented right after the serverTCP gets the request from Client. And those functions should be implemented before going to vector and respond to Client.
- How to get the entire package from Client Modbus? Because sometime I want to know how many registers the client requests, and the range of address (from 0-9 in this example, but sometime it will be from addr 5 to addr 14...), and maybe sometime I want to know the unitID in the client's package, currently I go to vector variable to get the unitID but I think that's not a good way.
So is there any ways to get all parameters in client's message: the range of address, unitID...

*P/S:* I'm a beginner in Node js, so maybe my questions are so basic, please kindly help me if you don't mind :D

Thanks for your time!

yaacov, 2017-08-08:
> Because I want other functions to be implemented right after the serverTCP got the request from Client

```
var doStuff = function(addr, unitID) { 
   // do a lot of interesting things before sending it to server

   return <the value to return to server>;
}

...
getHoldingRegister: doStuff,
...
```

> Because sometime I want to know how many registers the client requests, and the range of address (from 0-9 in this example, but sometime it will be from addr 5 to addr 14...)

Ooo... good point 👍 didn't think of that use case ... you are welcome to add this functionality ...

q-bird, 2017-08-14:
Hi,

I'm a beginner with nodeJs, so in order to fix this issue, I changed something in file `servertcp.js`:
```
for (var i = 0; i < length; i++) {
    responseBuffer.writeUInt16BE(vector.getHoldingRegister(address + i, unitID, address, length), 3 + i * 2);
}
```
The original is ` (address + i, unitID)`
I changed to `(address + i, unitID, address, length)`
The aim is to get the start address and to know how many registers the client requires.
And in the main file I can call: `var doStuff = function(addr, unitID, start_addr, num_regs) { ... }`

With this way I can get the `start_addr` and the `num_regs`, but the `doStuff` will repeat 'length' times because of the for loop, and of course the `start_addr` and `num_regs` will have the same value on those 'length' times.

So no need to repeat too much times to get `start_addr` and `num_regs` right? Is there any way to get those values at the first time before going to `doStuff` function?

Can you advise a better way?

Thank you!

yaacov, 2017-08-14:
> Can you advise a better way?

a. you should do a [pull request](https://help.github.com/articles/creating-a-pull-request/) :-)

b. you can add a new function to the vector  `getMultipleHoldingRegisters` , this function returnes an array (size length) of `UInt16` and you loop on this numbers to insert them to the `responseBuffer`

wuyuliyu, 2017-09-24:
[image]
Client power off restart

dushyantbangal, 2018-06-04:
@yaacov I'm doing a fix for this.
I'm also comparing the length of response with the number of requested values.
If these don't match, do I:

1. Throw an error message saying "Requested address length and response length do not match"
2. Just send the error to the callback which will respond with Exception 4
3. Both

commit ac109a37 (2018-06-05): Fixes #130
commit c52f1dfc (2018-06-05): Merge pull request #211 from dushyantbangal/master

Fixes #130
closed 2018-06-05 by yaacov