yaacov/node-modbus-serial #131: UART integrated Modbus-Serial

ercint, 2017-08-10:
Hi,

We would like to use your library but also we have to  set RTS signal while we are sending data to request response. So we will not use USB to RS485 converter, that is why we need to set it manually, in other way, we connect directly rs485 module from our device. 

So can you please tell us where should we set the bit in your package then clear it to wait the response?

Thanks in advance,
Regards

yaacov, 2017-08-10:
Hi,

a. see node node-serialport [1] for documentation about control signals.
b. feel free to update the node-serialport version we use [2] if you need more options.
c. the place to set the control bits is before and after send:
sending data:
https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L150
d. you will have to add a callback to know when to unset the bit:
be aware that when callback from `.write` is called dara "may not yet be flushed to the underlying port"
https://github.com/EmergingTechnologyAdvisors/node-serialport#serialportwritedata-encoding-callback--boolean

Thanks for the issue: waiting for your pull request :-)

[1] https://github.com/EmergingTechnologyAdvisors/node-serialport
[2] https://github.com/yaacov/node-modbus-serial/blob/master/package.json#L41

ercint, 2017-08-12:
> serialportwritedata-encoding-callback--boolean
> master/package.json#L41

Hi,

Thans for your reply, we are working on it but we cannot make callback work
after write function which you mentioned at line 150. Here is the part of
the code what we added.

Do you have any idea why callback is not called after write?

this._port.write(buffer, 'utf8', function(err){
                        console.log(err);
                        console.log("Callback Called");
                });

Thanks in advance

yaacov, 2017-08-12:
> Do you have any idea why callback is not called after write?

What port do you use ? 

Only the simple `connectRTU` [1] use serialport as is , if you use `connectRTUBuffered` you need to override the RTUBuffered `write` fuction, and add callback support here:
https://github.com/yaacov/node-modbus-serial/blob/master/apis/connection.js#L60
https://github.com/yaacov/node-modbus-serial/blob/master/ports/rtubufferedport.js#L169

have fun

[1] https://github.com/yaacov/node-modbus-serial/blob/master/apis/connection.js#L60

yaacov, 2017-08-12:
p.s.

a. maybe you should use the `flush` function callback ?
https://github.com/EmergingTechnologyAdvisors/node-serialport#module_serialport--SerialPort+flush

I did not try any (not write and not flush), but using flush looks more safe

b. some devices start sending response really fast after end of request, if you set the control bit too late, you may lose start of response, you should check for that too ( after you get the callback working :-) )

ercint, 2017-08-12:
Hi,

Thanks for your fast reply :) 

We are using RTUBuffered as you guessed. we make callback work but now it is returning timeout. 

We will check flush.

yaacov, 2017-09-24:
Closing as this is out of scope of this repository ...

closed 2017-09-24 by yaacov