yaacov/node-modbus-serial #132: Bug during the Modbus TCP Server running

q-bird, 2017-08-13:
Hi,

When I run the Modbus TCP, and I didn't touch anything, just see the result for a while, but sometime it shows the error as below... Then I have to re-run the program

Have you ever seen the same problem like this?

```
...node_modules/modbus-serial/servers/servertcp.js:105
        responseBuffer: responseBuffer.toString("hex")
                                      ^

TypeError: Cannot read property 'toString' of undefined
    at _parseModbusBuffer (.../node_modules/modbus-serial/servers/servertcp.js:105:39)
    at Socket.<anonymous> (.../node_modules/modbus-serial/servers/servertcp.js:391:34)
    at emitOne (events.js:77:13)
    at Socket.emit (events.js:169:7)
    at readableAddChunk (_stream_readable.js:153:18)
    at Socket.Readable.push (_stream_readable.js:111:10)
    at TCP.onread (net.js:537:20)
```

yaacov, 2017-08-13:
Hi, thanks for the issue !!

We will need to fix that :-) 

I will look at it next weekend, if you can investigate more, and maybe come up with a fix, it will be nice.

wuyuliyu, 2017-08-18:
Hi,

The erro below will appeared, when the network interface of client disconnect ,then connect it again.

events.js:182
      throw er; // Unhandled 'error' event
      ^
Error: read ECONNRESET
    at _errnoException (util.js:1022:11)
    at TCP.onread (net.js:610:25)

How to solve it ?  thanks :)

closed 2017-09-24 by yaacov