yaacov/node-modbus-serial #134: Modbus RTU Data Type

code19-io, 2017-08-28:
I have been using modbus-serial to communicate with some energy meters. Those Energy meters outputs 32 bit float values and sometime 64 bit double values. Are these data types are supported by this module ?

yaacov, 2017-08-28:
> Are these data types are supported by this module ?

You can read the raw buffer and parse the data yourself, here is an example of parsing `UInt32BE` from a buffer of two 16bit registers:
https://github.com/yaacov/node-modbus-serial/wiki/Examples:-clients#read-raw-buffer

IlhanSeven, 2017-08-28:
Whatever type the value is, data is stored in classic 16 byte Modbus registers. As yaacov said, you can get it raw and parse it accordingly. Just be careful about byte and word orders. It can change from meter to meter.

code19-io, 2017-08-28:
I'll try to get them as buffer and try

code19-io, 2017-08-29:
If i am using Modbus-Serial module to implement a modbus master and using the following settings to communicate with the modbus slave 

- BaudRate: 19200
- Databits: 8 bit
- Stopbits: 2 bit
- Parity: None
- RTSCTS: true

And using Modbus Slave (http://www.modbustools.com/modbus_slave.html) to simulate modbus slave and using virtual ports to communicate within my environment. 

Please help me on getting Binary information from the modbus slave without any conversion to any other data types including Integer double or float. Just the Binary Raw Data and how it is represented.

yaacov, 2017-08-29:
> Please help me on getting Binary information from the modbus slave without any conversion to any other data types including Integer double or float. Just the Binary Raw Data

See this example of parsing UInt32BE from a buffer of two 16bit registers:
https://github.com/yaacov/node-modbus-serial/wiki/Examples:-clients#read-raw-buffer

Data recived is a buffer, see buffer documentation for how to parse, and display it:
https://nodejs.org/api/buffer.html

yaacov, 2017-08-30:
> Thank you..... it worked finally.

:-)