yaacov/node-modbus-serial #149: serialport isOpen to Boolean - is not a function

biancode, 2017-09-30:
I did not see port.isOpen - [Serialport](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/test/serialport.js) line 219 test - is now Boolean not a function

biancode, 2017-09-30:
hint for me - Serialport should be wrapped to hold a static own API

biancode, 2017-09-30:
@yaacov what do you say - change isOpen to Boolean in that lib or build a wrapper for SerialPort to be compatible

yaacov, 2017-09-30:
> @yaacov what do you say - change isOpen to Boolean in that lib or build a wrapper for SerialPort to be compatible

I prefer to change isOpen to Boolean like:
https://github.com/EmergingTechnologyAdvisors/node-serialport/commit/5616664ed67a983a28aa54e08350f374e9be9085

yaacov, 2017-10-01:
@biancode hi,

We have a new version v6.0.1 that fixs this 🎉 🍰 
Can you check that it fixes this issue ?

biancode, 2017-10-01:
@yaacov - yes I'll check it tomorrow - thank you!

biancode, 2017-10-02:
@yaacov it works very well for me and for now, I'm very happy with that version

If you find some time try your Modbus server in Node-RED

[image]

yaacov, 2017-10-03:
> it works very well for me

closed 2017-10-03 by yaacov