yaacov/node-modbus-serial #162: Compatibility with windows ?

archgloom, 2017-11-07:
We can't use this on windows operating system.  Because it doesn't support setting serial path. 
Eg. in unix/linux we set serial port path from "/dev/ttyS0"  in Windows we have COM1,COM2,COM3  etc. 
So it does not work on windows NodeJS. 

what should we do to use it on windows ?

yaacov, 2017-11-07:
> what should we do to use it on windows ?

I did not try it myself, but a path like "COM1" should work.

Can you try using "COM0".."COM99" and describe the error or success ?

neldefant, 2017-11-08:
Works on Windows 10 (and Linux), node 6.10.0, using various USB-serial things.

// create a modbus client using the serial port
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

// Get configuration location
if (/^win/.test(process.platform)) {
   var port = "COM7";
}
else {
    var port = "/dev/ttyAMA0";
}

// open connection to a serial port
client.connectRTU(port, { baudRate: 19200, parity: 'even', stopBits:1, dataBits: 8, autoOpen:true}, function(err){ 
....
});

yaacov, 2017-11-08:
> Works on Windows 10 (and Linux), node 6.10.0, using various USB-serial things.

Greate 🎉

closed 2017-11-08 by yaacov