yaacov/node-modbus-serial #163: Make serverTcp handler async?

meisterlampe, 2017-11-08:
I think about getting the register values from an async api and I don't know how to block the synchronous function until the promise is resolved.

After a first look into the code (servertcp.js), it seems possible to make `_parseModbusBuffer` async and await all the handlers.
What's your opinion? Is there a reason for the tcp server functions to be sync?

yaacov, 2017-11-08:
hi, thanks for the issue, I will look at it this weekend ... ( hopefully :-) )

meisterlampe, 2017-12-07:
In my case I just needed an async call to `getHoldingRegister`, so I can do some async work in the handler.
You can find my changes in https://github.com/meisterlampe/node-modbus-serial/blob/master/servers/servertcp.js

I know, it is a hack breaking the API and far away from beeing a PR as all other methods should still need to be converted to async. Anyway it's a working test for me right now :)
Would you prefer another (maybe non-api-breaking) approach?

yaacov, 2017-12-07:
wow, thanks, I hope to have time to look at it ...

uwburn, 2018-01-11:
Hello,
i'm working on a callback based non API breaking version of this, as i needed to simulate some delay of the network for testing purposes.

If it works out ok, i will write up tests for it and submit PR. ;)

yaacov, 2018-01-11:
great 👍

yaacov, 2018-01-15:
@meisterlampe please check  #170

meisterlampe, 2018-01-16:
Hm, I tried the current master branch and it was not working for me.
Maybe I'm using this wrong.

My test:
1. Using Smply Modbus TCP Client 8.0.7 to read a 32bit Float (working with my async mod mentioned in this issue)
2. Modbus-Server sends `4369 3A32`
3. Receiving only the first UInt16. Complete result is e.g.: `4369 0000`
4. Reading another 32bit Float
5. From now on the **previous** results are returned, not the current. E.g. `4369 3A32`

What I changed in my code:
Added callback to my vector, that is passed to `new ServerTCP`-call
`getHoldingRegister: (addr, unitId) => self.getHoldingRegister(addr, unitId)`
result:
`getHoldingRegister: (addr, unitId, callback) => self.getHoldingRegister(addr, unitId, callback),`

getHoldingRegister has this implementation:
```
private async getHoldingRegister(addr, unitId, callback) {
    try {
      Logger.debug(`getHoldingRegister: addr:'${addr}', unitID:'${unitId}'`);

      let result = await this.holdingRegisterManager.getHoldingRegister(addr);
      Logger.debug(`Result: ${result}`);
      callback(result);
    } catch (error) {
      Logger.error("Error in getHoldingRegister", error)
    }
}
```

In my logs, I do see the Result-line for both requests. If I add some timeout before calling the callback of the second request, the test-client immediately shows the result of step 3 from my test description above, event though the second request has not yet called the callback.

I have no idea, what I've done wrong. The whole code is working with my async modifications from this issue (without the callback lines added :) ). It seems, the modbus server is not waiting for the second callback to be called.

yaacov, 2018-01-16:
@uwburn ^^

uwburn, 2018-01-16:
Ok, this is embarassing, i messed up the `servertcp.js` file as i started to work (in a hurry) on an older version and then copy-pasted it failing to merge it correctly.

I went through the commit history of the to check out that is now restored properly. I have committed it again on my fork. @yaacov do you want another PR?

@meisterlampe 
I'm not sure if that's the root of the problem, but `async/await`-`Promises` and `callbacks` are alternative approaches, you should use one of the two.

My modification does check the number of `getHoldingRegister` function parameters, if 3 it works with callback, if 2 it check for returned value; on the returned value it checks if it has a `then` property (meaning that's a `Promise`) and if it does it `then` is invoked resolving the result from it. As async functions, unless awaited, return Promises it should work on async/await too.

In your case, as you have 3 parameters, callbacks should win, and as you're invoking the callback properly passing result, i'm not really sure of what's going on.

Anyway, the proper implementation with `async/await` should be:
```
private async getHoldingRegister(addr, unitId) {
    try {
      Logger.debug(`getHoldingRegister: addr:'${addr}', unitID:'${unitId}'`);

      let result = await this.holdingRegisterManager.getHoldingRegister(addr);
      Logger.debug(`Result: ${result}`);
      return result;
    } catch (error) {
      Logger.error("Error in getHoldingRegister", error)
    }
}
```

if that doesn't solve, I would suggest:

1. Wait for a fixed version with the messed servertcp.js cleaned up, to exclude other causes
2. If possible, follow the execution with an async capable debugger

If you can't do step 2 but you can provide a sample client/server reproducing the bug i can look at it. :)

yaacov, 2018-01-16:
> @yaacov do you want another PR?

@uwburn sure :-)

yaacov, 2018-01-17:
@meisterlampe, @uwburn  please check #172 ;-)

meisterlampe, 2018-01-17:
Disconnecting is now handled well :)

I debugged my test app and I discovered the following behaviour:
- The Simply Modbus Client waits until the callback function is called the first time (or until the timeout reached, when I did not hit F5 fast enough ;) )
- After calling the callback the first time, the Modbus client immediately shows the result showing only the first UInt16 part. It is not waiting for the second part at all.

Maybe I will try to switch to callback-only, but this will take some time.
So if this works for you even for combined requests like getting an 32bit-float, this issue can be closed.

meisterlampe, 2018-01-17:
I now had a deeper look into the code and maybe I found the issue:

In the old version of `_handleReadMultipleRegisters` the buffer was filled for every requested part:
```
// read registers
if (vector.getHoldingRegister) {
   for (var i = 0; i < length; i++) {
        responseBuffer.writeUInt16BE(await vector.getHoldingRegister(address + i, unitID), 3 + i * 2);
   }
}
```
The buffer was only send to the client after it has been completely filled.

In the new code, a callback is called for every requested part:
```
responseBuffer.writeUInt16BE(value, 3 + i * 2);

modbusSerialDebug({ action: "FC3 response", responseBuffer: responseBuffer });

callback(responseBuffer);
```

The callback writes to the responseBuffer and then calls its own callback `callback(responseBuffer);` even if not all data has been written to `responseBuffer`.
But calling the callback leads to writing to the socket, which sends the response to the client even though the response is not complete yet.

A solution could be to count the calls to the callback-function created by `buildCb(i)` on line 217 and only call the callback-parameter of `_handleReadMultipleRegisters`, when it has been called the "length"-times.

What do you think? (I hope I haven't missed something O:-) )

yaacov, 2018-01-17:
> A solution could be to count the calls to the callback-function created by buildCb(i) on line 217 and only call the callback-parameter of _handleReadMultipleRegisters, when it has been called the "length"-times.

@uwburn WDYT ?
@meisterlampe , can you fix it, so it will work for you, and send a pull request ?

meisterlampe, 2018-01-17:
Created a PR which fixes getHoldingRegister for me.

yaacov, 2018-01-17:
@uwburn can you check if this works for you (https://github.com/yaacov/node-modbus-serial/pull/173) ?

uwburn, 2018-01-17:
@meisterlampe bingo, i think your explanition is correct!

Looking at #173, there are other cases other than `_handleReadMultipleRegisters` where the `buildCb` approach is used. Also, i think `i` and `countCallbackCalls` should be the same, so you can just check for `i === length`.

yaacov, 2018-01-17:
> there are other cases other than _handleReadMultipleRegisters where the buildCb approach is used. Also, i think i and countCallbackCalls should be the same, so you can just check for i === length

@uwburn can you make the fixes in a pull request ?

yaacov, 2018-01-17:
https://github.com/yaacov/node-modbus-serial/pull/174 ready for re-testing  

@uwburn @meisterlampe can you re-check ?

meisterlampe, 2018-01-17:
I would say, checking for i === length is not enough.
You cannot tell in which order callbacks will be called, because the invocation of the callback is async. Therefore, the callback for i === length must not necessary be the last one, that is called.

yaacov, 2018-01-17:
> because the invocation of the callback is async.

so if we use a counter (like in #173) it's will work for all @meisterlampe @uwburn ?
my vote is for using a counter if it doesn't break anything else :-)

uwburn, 2018-01-17:
I see your point: considering that a dedicated counter is definitely needed. ;)
I'm off for tonight, but i can update the PR tomorrow.

yaacov, 2018-01-17:
> but i can update the PR tomorrow.

great !

uwburn, 2018-01-18:
Submitted. :)

Just to be sure, callbacks resolution order is not important as `responseBuffer.writeUInt16BE(value, 3 + i * 2)` (and similar) is called with an offset, correct?

yaacov, 2018-01-18:
> Just to be sure, callbacks resolution order is not important as responseBuffer.writeUInt16BE(value, 3 + i * 2) (and similar) is called with an offset, correct?

a. yes, `writeUInt16BE` is always called with an offset.
b. @meisterlampe @uwburn last round of checks, just to be sure ?

yaacov, 2018-01-18:
referencing the pull request: https://github.com/yaacov/node-modbus-serial/pull/175

meisterlampe, 2018-01-18:
Looks good to me and this is working in my application without problems.
(I'm using only getHoldingRegister, but the other handlers seem to be very similar and they look good to me too)

uwburn, 2018-01-18:
I was thinking about error handling: a proper callback implementation should handle an error first parameter and promises should be catched.

For me this is not particoularly relevant, as i'm using the server for testing purposes, it never throws errors, as it answer mostly fixed result, but if the server is meant to deal also with real devices i think that would be needed.

Dealing with errors should be rather easy, what i'm missing out is what the server should respond to the client when the error happens.

yaacov, 2018-01-18:
> this is working in my application without problems

Thanks 👍 

For me it was a lot of fun to do this together ! thanks @meisterlampe and @uwburn 

> Dealing with errors should be rather easy, what i'm missing out is what the server should respond to the client when the error happens.

If one day you decide to implement this, just submit a pull request :-)

uwburn, 2018-01-18:
> this is working in my application without problems

Good!

As for the error handling, i have it mostly implemented by now... i need to review the changes a little further before submitting PR again, i will do it in a day or so.

In case of an error i'm doing
```
if (err) {
    modbusSerialDebug({
        error: "error processing response",
        unitID: unitID,
       functionCode: functionCode
    });

    var errorCode = 0x04; // slave device failure

    // set an error response
    functionCode = parseInt(functionCode) | 0x80;
    responseBuffer = Buffer.alloc(3 + 2);
    responseBuffer.writeUInt8(errorCode, 2);

    cb(null, responseBuffer);
    return;
}
```

meisterlampe, 2018-01-18:
Thank you for solving this issue that lightning fast @yaacov and @uwburn. It was fun for me too and still is :-)
@uwburn Error handling looks good to me. This enables me also to signal errors from the callback, which was not possible before in the sync version, right? That would be a new feature 👍

uwburn, 2018-01-18:
No problem at all, it's nice to contribute to this project! :)

As for the error handling, for sync and Promises it will be enough to `throw`, callbacks will need to be called passing a non-null first parameter.

uwburn, 2018-01-19:
I'm setting up a test case, and i found an unhandled condition: what should happen if `length` as in `_handleReadMultipleRegisters` is 0? Is it an error to do a 0 length request or should the response be sent anyway?

yaacov, 2018-01-19:
p.s.
we only call this functions if `length > 0` [1] so currently it will simply do nothing ...

[1] https://github.com/yaacov/node-modbus-serial/blob/master/servers/servertcp.js#L184

uwburn, 2018-01-19:
Yes, that's the problem, no callbacks would be invoked, ending up with the server never responding.

Submitted new PR anyway, if no problemi arises i think we're done.

BTW, i think async support in server also addresses #144.

yaacov, 2018-01-19:
@uwburn  added error handling and docs: https://github.com/yaacov/node-modbus-serial/pull/176

@meisterlampe , can you verify this doesn't break anything on your end ?

yaacov, 2018-01-19:
> BTW, i think async support in server also addresses #144.

Nice :-)

meisterlampe, 2018-01-19:
@uwburn You're right. With async support, the response delay feature from can easily be implemented handler-functions. I'd say #144 is resolved with this.

meisterlampe, 2018-01-19:
> @meisterlampe , can you verify this doesn't break anything on your end ?

I will check next week. Weekend ..hooray :)

yaacov, 2018-01-19:
@uwburn after @meisterlampe will give this an OK, and maybe we will here from @biancode, I will make a new version and publish it to npm ... 

Thanks all !!

biancode, 2018-01-19:
I'll test it.

yaacov, 2018-01-27:
@biancode any news ?

biancode, 2018-01-27:
I'm sorry, influenza stop ...

yaacov, 2018-01-28:
@biancode fill well soon :stew: (*)

(*) this is a soup imoji

yaacov, 2018-02-05:
Fixed by #170 .. #176

closed 2018-02-05 by yaacov

yaacov, 2018-02-05:
v7.1.3 published 🍰

biancode, 2018-02-05:
normal state works - now I start to test some heavy loads

yaacov, 2018-02-05:
> normal state works - now I start to test some heavy loads

great !