yaacov/node-modbus-serial #164: Read Multiple Registers at once

Athiththan, 2017-12-15:
I have been using this module to read data from Modbus devices, and now I am in need of reading 3 different registers from the same device. How should i do it ??? 

Thanks in advance.. (still a beginner)

yaacov, 2017-12-15:
Hi, thanks for the issue :-)

a. if this registers are near, you can use one call to readInputRegisters.
for example, if you want to read registers 2, 5 and 7, you can read 6 registers starting from register 2:
`client.readHoldingRegisters(2, 6)`

b. if this registers are far apart, you can use several call to readInputRegisters.
for example, if you want to read registers 2, 500 and 501:
`client.readHoldingRegisters(2, 1)`
`client.readHoldingRegisters(500, 2)`

Athiththan, 2017-12-15:
OK thank you for the quick response .... 

I am having the Part B issue and as well as i have to read them with a timeout... how should i achieve that without conflicting with any others 

Thanks in advance

yaacov, 2017-12-15:
You can use promises or callbacks:

```
function nextRead() {
   client.readHoldingRegisters(5, 1)
      .then(continueProgram)
   ...
```

Using premisses:
```
client.readHoldingRegisters(5, 1)
   .then(nextRead)
```
Using callbacs:
```
client.readHoldingRegisters(5, 1, nextRead)
```

yaacov, 2018-02-10:
I think this is closed :-)

closed 2018-02-10 by yaacov