yaacov/node-modbus-serial #165: Blue Screen Windows 10

sacha67, 2017-12-15:
Hello,

I'm using your module to comunicate with a device using RTU connection.

I tried to use your functions with async/await and had multiple windows Blue Screen Errors ("system thread exception not handled" or "multiple irp complete requests windows").

`	async testFunc(): Promise<any> {
		const client = new ModbusRTU();
		await client.connectRTUBuffered('COM4');
		client.setID(3);
		client.setTimeout(5000);
		const readRegisterResult: ReadRegisterResult = await client.readHoldingRegisters(50000, 66);
		client.close();
		return Promise.resolve(readRegisterResult);
	}`
Is your module handling this type of code well or should I only use callback functions only ?

yaacov, 2017-12-15:
Hi, thanks for the issue,

I do not have a windows machine ... I can't help you here. From the error it sounds like a low level promise/node issue, do use the last node version ?, If you find a way to solve it, please leave a comment here.

sacha67, 2017-12-15:
I have node version: 8.9.1

yaacov, 2017-12-15:
Current is 9.3.x , you may try to upgrade ...

yaacov, 2018-02-05:
Hi, since I do not have a windows machine to check this, and this was raised on an old version, closing ...

closed 2018-02-05 by yaacov