yaacov/node-modbus-serial #166: No support for v8 node.js

dom-white, 2018-01-05:
Sometime ago an 'engine' limitation was added to this repo to prevent node-modbus-serial running on anything other than node.js v4 or v6.

As of October 2017, v8 is the latest LTS version of node.js

Is it possible to add v8 to the list of acceptable engines?

Ideally I would remove this restriction as it is just another thing that requires maintaining/updating. Plus I am not sure why such a strict enforcement policy is really required. I run v7 currently (for various reasons, with a plan to move to v8 shortly) and so I would not be able to use the latest version of modbus without forking and patching.

Wouldn't a 'tested with version x and y' statement in the readme suffice?

Thanks
Dominic

yaacov, 2018-01-06:
> Is it possible to add v8 to the list of acceptable engines, or to remove the limitation?

my vote - remove the limitation :-) 

> Wouldn't a 'tested with version x and y' statement in the readme suffice?

my vote - yes

can you make a pull request removing this check, and add some notice in the readme ?

yaacov, 2018-01-08:
fixed by #167

closed 2018-01-08 by yaacov