yaacov/node-modbus-serial #171: Modbus RTU over TCP/IP

sergiupienar, 2018-01-12:
Hi, 

I`m using the Modbus Slave emulator (from modbustools) that is set in RTU over TCP/IP mode and can't manage to get data out of it when connecting with modbus-serial.

I can see there is distinction when it comes to the Rx made from an app that works well (Modbus Poll) and the calls that I am making from modbus-serial.

When using Modbus Poll (the app that works) the Rx is headless and has a CRC added to it, as expected, e.g.:

Rx:000010-01 01 00 00 00 04 3D C9
Tx:000011-01 01 01 00 51 88

while when I`m calling from node the RX looks like one that's being sent over plain TCP:

Rx:000000-00 01 00 00 00 06 01 03 00 00 00 04

and the app will timeout.

The test code I`m using is pretty much the same you've posted in one of the examples:

```var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

function close() {
    client.close();
}

function run() {
    client.readHoldingRegisters(0, 4)
        .then(function(d) {
            console.log("Receive:", d.data);
        })
        .catch(function(e) {
            console.log("Exception " + e);
        })
        .then(close);
}

function setClient() {
    // set the client's unit id
    // set a timout for requests default is null (no timeout)
    client.setID(1);
    client.setTimeout(2000);
    // run program
    run();
}

client.connectTcpRTUBuffered("127.0.0.1", { port: 502 })
    .then(setClient)
    .then(function() {
        console.log("Connected");
    })
    .catch(function(e) {
        console.log("Exception: " + e);
});```

yaacov, 2018-01-12:
Hi, Thanks for the issue !

I think it's a documentation bug, IIRC the `connectTcpRTUBuffered` is the same as `connectTCP` but with frame check and concatenation of broken packages (for example, esp8266 may break packages from time to time).

I think that `connectTelnet` is closer to what you are looking for.

If the telnet port works for you, please comment here so I can fix the description of `connectTcpRTUBuffered` to be more precise.

sergiupienar, 2018-01-12:
Hi,

Indeed, connectTelnet worked well when the Modbus Slave is set to RTU over TCP/IP.

Thanks for the hint.

closed 2018-01-12 by sergiupienar

mohammed-papad, 2022-04-13:
thanks connectTelnet works