yaacov/node-modbus-serial #179: Modbus exception 4 with 7.1.3

biancode, 2018-02-06:
Hi @yaacov ,

[image]

all as before but now exception ... it is a test with positive and negative values

[image]

I will test more the next few days, but maybe you see some to give a hint ...

cya
Klaus

biancode, 2018-02-06:
FC5 and FC16 works well, but FC6 doesn't

yaacov, 2018-02-07:
maybe:
https://github.com/yaacov/node-modbus-serial/blob/master/servers/servertcp.js#L510

should be:
`callback(null, responseBuffer);`
instead of:
`callback(responseBuffer);`
?

biancode, 2018-02-07:
Yes, that is the problem.

yaacov, 2018-02-07:
fixed by #180

yaacov, 2018-02-07:
@biancode 7.1.4 is out 🎆

closed 2018-02-07 by yaacov