yaacov/node-modbus-serial #182: Auto-reconnect?

robertsLando, 2018-02-07:
Haven't found any information about this, is auto reconnect enabled or shuld it be self made?

yaacov, 2018-02-07:
hi, thanks for the issue,

I think you need to check the errors and do reconnect manually if needed.

@biancode do we need self heeling here ?

biancode, 2018-02-07:
@yaacov for me self-healing is not the best because mostly there is some to be done before reconnecting or you have to wait a specific time or you have to check some before ...

yaacov, 2018-02-07:
> for me self-healing is not the best

@biancode thanks, this is what I think too.

@robertsLando you need to catch the relevant errors and reconnect.

biancode, 2018-02-07:
libraries like jsmodbus provide self-healing and queueing and more ...
but that is what I love about modbus-serial - it is more a Modbus API then an all you can eat lib

yaacov, 2018-02-07:
@robertsLando the examples dir have examples of how to catch errors.

Here is an example with debug example that will help you find more info about errors:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/debug.js

closing ?

robertsLando, 2018-02-07:
Ok thanks @yaacov and @biancode .

robertsLando, 2018-02-07:
@yaacov  sorry I forgot to ask: HOw to catch errors? Is there events handlers? client.on('error')?

yaacov, 2018-02-07:
here is an example:
```javascript
client.readHoldingRegisters(5, 4)
        .then(function(d) {
            console.log("Receive:", d.data); })
        .catch(function(e) {
            console.log(e.message); })
.then(close);
```
more examples here:
https://github.com/yaacov/node-modbus-serial/tree/master/examples

yaacov, 2018-02-07:
you can also use debug mode and see more data about why things fail, and handle it more intelligently:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/debug.js

yaacov, 2018-02-07:
ahh ... and also catch connection errors:
```javascript
client.connectTCP("127.0.0.1", { port: 8502 })
    .then(setClient)
    .then(function() {
        console.log("Connected"); })
    .catch(function(e) {
        console.log(e.message); });

```

robertsLando, 2018-02-07:
Thanks but what if connection drops for any other reason? Should I check that connection is still open with `client.isOpen` and in the case isn't try to reconnect or I get the error from the read and than decide if reconnect or not? This is why I was asking for client events like 'onError' 'onConnected' 'onClose'.

yaacov, 2018-02-07:
> Thanks but what if connection drops for any other reason?

the `.catch` of the `.connectTCP` should be called.

> Should I check that connection is still open with `client.isOpen`

if `client.isOpen` is false the `.catch` in `client.readHoldingRegisters` should be called.

> and in the case isn't try to reconnect or I get the error from the read and than decide if reconnect or 
not? 

you should imnplement the `.catch` of `.connectTCP` and  `client.readHoldingRegisters`.

> This is why I was asking for client events like 'onError' 'onConnected' 'onClose'.

this will all call `.catch` of the `.connectTCP` with different `error` objects.

closed 2018-02-09 by yaacov