yaacov/node-modbus-serial #184: Uncaught TypeError: Cannot redefine property: isOpen

robertsLando, 2018-02-09:
Code: 

```
if(self.client && !self.client.isOpen)
    self.client.open(onConnect.bind(self));
  else{
    var client = new ModbusRTU();

    if(self.info.tcp)
    client.connectTCP(self.info.address, { port: self.info.tcp_port }, onConnect.bind(self));
    else
    client.connectRTUBuffered(self.info.port, { baudRate: self.info.baud_rate }, onConnect.bind(self));

    self.client = client;
  }
```

```
Uncaught TypeError: Cannot redefine property: isOpen
    at Function.defineProperty (<anonymous>)
    at ModbusRTU.open (/media/daniel/DATA/MQTT_Gateway_Node/node_modules/modbus-serial/index.js:349:12)
    at setClient (/media/daniel/DATA/MQTT_Gateway_Node/modbus-reader.js:84:17)
    at Timeout._onTimeout (/media/daniel/DATA/MQTT_Gateway_Node/modbus-reader.js:101:25)
    at ontimeout (timers.js:386:14)
    at tryOnTimeout (timers.js:250:5)
    at Timer.listOnTimeout (timers.js:214:5)
ModbusRTU.open @ index.js:349
setClient @ modbus-reader.js:84
(anonymous) @ modbus-reader.js:101
ontimeout @ timers.js:386
tryOnTimeout @ timers.js:250
listOnTimeout @ timers.js:214
```

yaacov, 2018-02-09:
Hi,

Thanks fot the issue, but without knowing what you I are trying to do it's hard to help.
a. Are you trying to redefine `isOpen`, why ?
b. why do think `onConnect.bind(self)` return a function that can be called with error argument on connection faill ?

robertsLando, 2018-02-09:
I'm not tring to refefine isOpen anywhere, I don't understand the error. I have created a class called ModbusReader that handles some modbus read so the `self` and `this` refers to this class. Before reading I check that connection with client is open, if is not open I call the method reconnect that calls setCLient that checks is there is an existing client and if is not opened it try to reopen the connection.

```
/**
 * Init Modbus Client
 */
function setClient(self){

  if(self.client && !self.client.isOpen)
    self.client.open(onConnect.bind(self));
  else{
    var client = new ModbusRTU();

    if(self.info.tcp)
    client.connectTCP(self.info.address, { port: self.info.tcp_port }, onConnect.bind(self));
    else
    client.connectRTUBuffered(self.info.port, { baudRate: self.info.baud_rate }, onConnect.bind(self));

    self.client = client;
  }
}

/**
 * Reconnect function
 */
function reconnect(self){
  setTimeout(function(){setClient(self)}, TIMEOUT_RECONNECT);
}

/**
 * Function called when Modbus client connects or an error occurs
 */
function onConnect(err) {
  if(!err){
    console.log("Modbus client connected: " + this.info.clientName);
    this.client.setTimeout(TIMEOUT_REQUEST);
    this.startScan = Date.now();
    startReading(this);
  }else {
    console.log(err);
    console.log("Retry in " + TIMEOUT_RECONNECT/1000);
    reconnect(this);
  }
}

```

robertsLando, 2018-02-09:
Maybe I have understand why this happens. In index.js line 349 you have defined isOpen property inside open method. So when the method is called the second time you try to redefine isOpen property that was already defined.

robertsLando, 2018-02-09:
Ok I have fixed this. Just move the define property outside the `.open` method and replace `this` with ModbusRTU.prototype.

```
/**
 * Check if port is open
 */
Object.defineProperty(ModbusRTU.prototype, "isOpen", {
    enumerable: true,
    get: function() {
        if (this._port) {
            return this._port.isOpen;
        }

        return false;
    }
});

```

robertsLando, 2018-02-09:
#185

yaacov, 2018-02-09:
I have tested:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger.js

that check for errors and re-connect if needed, and it works fine without the fix, why do you need this fix ?

yaacov, 2018-02-09:
merged #185 👍 

p.s.
a. I do think you need to check what you are doing more carefully.
look at how the logger example handles the disconnect error, maybe you will find a cleaner way to do what you want.
b. you should give a link to your projects github page, maybe someone can help fix that error :-)

yaacov, 2018-02-09:
fixed by #185

yaacov, 2018-02-09:
v7.1.5 with #185 is published on npm

Thanks @robertsLando 👍

closed 2018-02-09 by yaacov

robertsLando, 2018-02-09:
> a. I do think you need to check what you are doing more carefully.
look at how the logger example handles the disconnect error, maybe you will find a cleaner way to do what you want.

@yaacov what do you mean? Sorry, I don't understand :(

yaacov, 2018-02-09:
> Sorry, I don't understand :(

the logger.js example does what you are trying to do without using `isOpen` or `bind` , if you need to use this functions you may be doing something wrong ( or not ... ) I didn't see your complete code :-)

robertsLando, 2018-02-12:
@yaacov I need to bind the callback function to the context of the main class to be able to access the class attributes once the callback function is triggered. I could did this without using `.bind` just by writing the function locally like `.connect(function(err){})` but as I use it more times I have choose to create a private function inside the class.

In logger example you check the error than you close the port and you recreate one but why to close the port and create a new one when you can just call `.open()` method whit the already existing client? I do the reconnect in this way and it works.